gnea/grbl-Mega #3: ATMega16U2 virtual com port on ATMega2560

Lob0426, 2016-09-03:
We have spent three days trying to get consistent communications with a Mega2560R3. The problem was the ATMega16U2. We could not consistently get welcome messages from the Mega2560 board. We found that cutting the trace so it would not Auto Reset has fixed most of these problems. Tried XON/XOFF and several other settings first. Some of these helped, but cutting the reset trace looks like the thing to try!

Trying to connect the Mega2560 to a Eleks Maker L6 or L7 board through the headers after the Nano has been removed. This is for the Chinese Benbox DIY kit laser engravers. then try to make a "Shield" that is specific to our laser machines after proving the concept and performance of the Mega2560.

chamnit, 2016-09-03:
@Lob0426 : Not sure if this is a bug report or just a comment. As for the atmega16U, this is the first time I've heard of any communication problems with those chips in particular. It's much more common to hear about the CH380 USB-serial chips having issues. Depending on what kind of comm error you've been getting, it could be related to EMI.

chamnit, 2016-09-03:
@Lob0426 : See this thread (https://github.com/grbl/grbl/issues/995).

Lob0426, 2016-09-04:
I had already followed the DFU thread and updated the 16U2
https://www.arduino.cc/en/Hacking/DFUProgramming8U2
and we were still having problems with a program called BenCutLaser on benboxlaser.us. Still seeing problems as grbl settings do not come in one piece they are broken up. Tried an Uno and had better responses. After we cut the trace it got better but still some problems. It is not grbl as far as I know but the Mega. I have the FLIP drivers installed. Others have had the same problem with the Mega board and BenCutLaser. T2Laser also was acting up until the trace was cut now it is operating on my Mega board.

Now If I could get the Laser to work under board control I would be almost there! The jumper under the Nano is throwing me off as to the connections for the laser to work under PWM control. Instead laser stays on with no control. Should be Mega D6 to Eleks Maker board D12 (jumper in 0.8 pos.). but not working so far

Lob0426, 2016-09-11:
UPDATE:
We found our problem was with the software. It was sending data to the board immediately after it connected. There needs to be at least a 1 second wait before data is sent or the Mega will wait for a sketch to be sent. This will either delay the welcome message or it will never appear.

https://www.arduino.cc/en/Guide/Troubleshooting#toc19
see link; Why doesn't my start when I power up or reset my Arduino board?
Not a grbl issue but it has delayed us in our grbl-mega project for some days!

closed 2016-09-14 by Lob0426