gnea/grbl-Mega #14: error messages codes

MoonshineSG, 2017-01-01:
what happened to the error descriptions ? in the previous versions the error messages (and the settings) has nice explanations, now it's all gone... :(

chamnit, 2017-01-01:
@MoonshineSG : I thought I had a FAQ answer for this, but I guess not. It was removed because of the flash limits of the 328p. It was either the human-readable messages or overrides/laser mode. Not both. I may bring it back in the Mega version soon, but it's not a priority. GUIs have been instructed to supply this information for the user. There is a set of CSV files in the repo for them to parse and keep updated from.

closed 2017-01-01 by chamnit

MoonshineSG, 2017-01-02:
Thanks for pointing to the cvs files. it helps.