gnea/grbl-Mega #16: pinout same as arduino uno?

f0m3, 2017-01-18:
_This is my first Issue ever so please correct me if i am doing it wrong!_
I compiled to mega version and uploaded to an arduino mega. I can communicate with grbl by universal gcode sender. But my old setup (i made a shield for my old arduino uno) doesnt work with the mega. Are the pinouts the same? If not this maybe should be mentioned on the wiki.

chamnit, 2017-01-18:
@f0m3 : The pin out is different. A lot different. See the cpu_map.h file for wiring. There was a picture somewhere showing the pin layout, but it was never added to the Wiki.

closed 2017-01-18 by chamnit

fabianmoronzirfas, 2019-06-06:
There is an issue for the pinout on the (old) original repo. https://github.com/grbl/grbl/issues/390. Would be great to have the images on the README

109JB, 2019-06-06:
> There is an issue for the pinout on the (old) original repo. grbl#390. Would be great to have the images on the README

There already is.

https://github.com/gnea/grbl-Mega/wiki/Connecting-Grbl-Mega

fabianmoronzirfas, 2019-06-06:
Great. I didn't look into the wiki.