gnea/grbl-Mega #17: grbl port on xmega avr

shamsiqbal, 2017-02-01:
hi
i have been working on grbl port on xmega controller i have seen an older port of grbl to xmega avr but thats V0.8 grbl port 
i have added many new features as  i have more processing power in xmega avr 
at this moment i have few question 
how to configure xmega uart for communication , how does serial communication works  in grbl current release i am working on v1.0 version here i want to know that does serial communication use any "print" c library function or it uses only interrupt  driven communication 
if it uses  only interrupt  driven communication then it will be easier to configure xmega interrupts and i am good to go with it 
second question is ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING how does this works in real time and how to adopt this  feature in xmega avr timer 
waiting for quick guide

chamnit, 2017-02-01:
@shamsiqbal : The source code is very well commented. Please start by reading and understanding it. Second, Grbl is being ported to 32-bit ARM.

closed 2017-02-01 by chamnit

shamsiqbal, 2017-02-01:
@chamnit : i was expecting to get answer that it uses "print ,printf ,prints" c library routine or not in any part of grbl code i have already added new parameters in mega2560 v1.0 for closed loop servo motor  control added pid closed loop control function now i am just one step ahead from testing it on my cnc milling machine about ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING this part was not implemented in older port of xmega avr so i asked it i read the whole code indeed its well commented thats why i was able to add above stated function in it i hope you will guide me little about it 
i after initial testing i will go to add missing G33 (tapping / threading ) function in it