gnea/grbl-Mega #19: Doxygen?

tgit23, 2017-02-25:
I was in the middle of regex'ing Grbl-Mega code comments for code documentation with Doxygen on a fork;  but realized later I cannot publish these on my local 'forked' repository as far as I could determine.  Please let me know if this incorrect.

So; I've copied the work into a non-forked repository and published it at https://tgit23.github.io/grbl-Mega-Doxy/index.html

I'd be willing to work on this further if it would be considered on a pull request.  It would add a /docs directory to the repository and it may have to publish clear back on grbl/grbl (not sure on those details as it appears the grbl-Mega is also a 'fork'.

Anyways; thought I'd open this issue for discussion.

chamnit, 2017-02-25:
@tgit23 : Thanks for sharing. At the moment, I'm working on overhauling significant parts of the code into a HAL. This basically means I'm moving on from how things are written in Grbl, because I had to make some unsavory coding to make things fit into the 328. This new code will be introduced as a new project and I would welcome documentation help there when it's released. I'm getting close to done but it's not available to the public until it is.

tgit23, 2017-02-25:
This is GREAT NEWS!  The reason for the doxygen changes was actually a distraction just to grasp a better understanding of the code so I could implement uni-polar stepper support in the best possible way.  Sounds like that's going to be easier if not already on its way with a HAL.

I actually had started approaching this by adding lines in cpu_map.h
#define STEP_PORT_SET(i) STEP_PORT=i
#define DIRECTION_PORT_SET(i) DIRECTION_PORT=i

and going through the 'stepper.c' code and doing a find/replace on lines such as:
STEP_PORT_SET((STEP_PORT & ~STEP_MASK) | st.step_outbits);

I guess I was attempting to abstract the PORT addressing;  but if this is something your current work will make much easier - I'd just as soon wait for it.  My email is on my github profile page @ https://github.com/tgit23 if you'd be interested in sharing what you have thus far; just out of my own curiosity

Also the changes to the code comments and the minor 'beginning' changes which I'd made are on the same repository as the doxygen publication source @ https://github.com/tgit23/grbl-Mega-Doxy just in case you wanted to check out the format for code comments I'd implemented. ( 1/2 done of course )

closed 2017-07-18 by chamnit