gnea/grbl-Mega #25: jog after probe problem

langwadt, 2017-05-28:
I'm running the code on an STM32 so it entirely possible I screwed something up porting, but before 
I start digging I'd like to know if it is a common issue. 

The first jog following a probe, set the state to "run" rather than "jog" which means jog cancel is ignored
once it's back to idle any following jog works as expected, setting the state to jog and respecting cancel

A-J-Bauer, 2018-12-17:
I can confirm this. I also have to issue a Cycle Start / Resume after probing to make jogging work as expected again (machine state JOG instead of RUN).

fra589, 2021-09-26:
Hi all,
I found the initial cause of this (old) bug :smile: 
The sys.step_control was not restored after the end of the probe cycle in motion_control.c because the probe cycle is initialized in the motion_control.c file and not in the protocol.c file.

The correction consist of adding this line after the line 306 of motion_control.c:
`sys.step_control = STEP_CONTROL_NORMAL_OP; // Fix #25 issue: restore step control to normal operation`

@chamnit
I can't make a pull request since my own fork grbl_Mega-5X is too far away of this version.
If you want, I added the diff file of motion_control.c (motion_control.diff.txt) so you can use it to fix your version.

[motion_control.diff.txt](https://github.com/gnea/grbl-Mega/files/7231113/motion_control.diff.txt)

@++;
Gauthier.