gnea/grbl-Mega #30: Mega for heavy duty engraving laser job

mkeyno, 2017-07-09:
hi , 
with enhanced memory of Mega2560 how much serial speed can be increased and what is the maximum stable buffer for  heavy duty CNC  job like laser  engraving ?
BR

mayhem2408, 2017-07-09:
@mkeyno I found that the larger buffers do help a bit, but at the end of the day, the serial bandwidth and processing power is the same. So the speed increase is noticeable but not massive.

If you want serious speed increases, try grbl-lpc on a smoothieboard. It has a planner buffer that is 15x larger, a serial buffer that is 32x larger and a much faster 32 bit MPU. I have sustained Feeds of 10000mm per minute doing greyscales., 12000mm per minute doing dithering and a massive 30000mm per minute doing black and white linearts.

chamnit, 2017-07-09:
@mayhem2508 : CPU speed and buffer sizes(actually can hurt you if too big) are only part of the bottlenecks. There are other improvements I plan to make that will remove the main one. It'll likely make the Mega speeds double and the LPC version even faster with engraving jobs. 

We'll see how much time I'll have to work on it. It's not that difficult to implement. It probably won't appear until the next major version is released later this year or early next. One to 1.5 year release cycles seem to be what I can handle these days.

mayhem2408, 2017-07-09:
@chamnit Well we you do implement these other improvements, I'm on board for testing.

mkeyno, 2017-07-11:
thanks @mayhem2408  actually I have had this discussion with @chamnit  before, but as he said it has not implemented yet,  and right now I could supply the Arduino  only , and my test shows the engraving speed is disappointing even with laserweb4 optimized gcode generator, however I've trying to build new controller with STM32F103 and Mega2560 to check how new boards could be different in practice 

@chamnit  I'm dying to see your new release repo with base 64 raster gcod implementation

chamnit, 2017-07-11:
@mkeyno : I've never said anything about a base64 raster gcode implementation. I've seen Marlin's description of it and don't agree with it. Bandwidth is not the primary bottleneck and Base64 will only marginally help.

closed 2017-07-11 by chamnit