gnea/grbl-Mega #33: Z-Enable pin on RAMPS 1.4 collides with CONTROL_RESET pin

urishab, 2017-07-31:
I'm trying to run the GRBL-Mega firmware on a RAMPS 1.4 compatible board.
X and Y are running well but Z does not.
I've found both Z-Enable PIN and CONTROL_RESET PIN are set to the same PIN:
PK0 (digital 62 I think) and this prevents the Z from working.
If I change the CONTROL_RESET pin to PK1 the Z axis works.

I'm guessing that I missed something that's supposed to disable the CONTROL_RESET, am I right?

In any case it seems that whatever it is, the pin should be disabled in RAMPS configuration

sergioausrio, 2017-08-01:
I have the same Issue on Ramps 1.4 
In which file i have to change the PIN Settings ?

chamnit, 2017-08-01:
I've shifted the control pins by plus one so that they are all on the RAMPS AUX 2 port, which should be unused. I don't have much time these days, so I'm depending on user feedback to find issues like this one. Please give it a try and let me know it works.

closed 2017-08-01 by chamnit

sergioausrio, 2017-08-01:
It works now. Thank You.