gnea/grbl-Mega #39: Spindle PWM non-functional

MaxBareiss, 2017-09-23:
I am using this version of grbl-mega with a RAMPS 1.4 board:
```
$I
[VER:1.1f.20170802:]
[OPT:VNML,35,255]

$$
$0=10
$1=100
$2=0
$3=2
$4=0
$5=0
$6=0
$10=1
$11=0.020
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=0
$24=500.000
$25=2000.000
$26=250
$27=5.000
$30=255
$31=0
$32=0
$100=200.000
$101=200.000
$102=4535.000
$110=3000.000
$111=3000.000
$112=300.000
$120=400.000
$121=400.000
$122=55.000
$130=600.000
$131=600.000
$132=60.000
```

I don't get any spindle PWM. If I turn the spindle on with any value but zero (`M3 S1`) I get full on, and when I turn it off with `M5` or `M3 S0` it turns off, but I don't get any PWM. I've modified this section of `cpu_map.h`:

```
//Control Digital Pin 6 which is Servo 2 signal pin on Ramps 1.4 board
  #define SPINDLE_TCCRA_REGISTER    TCCR4A
  #define SPINDLE_TCCRB_REGISTER    TCCR4B
  #define SPINDLE_OCR_REGISTER      OCR4C
  #define SPINDLE_COMB_BIT          COM4C1

  // 1/8 Prescaler, 16-bit Fast PWM mode
  #define SPINDLE_TCCRA_INIT_MASK ((1<<WGM40) | (1<<WGM41))
  #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41) | (1<<CS40)) 
  #define SPINDLE_OCRA_REGISTER   OCR4C // 16-bit Fast PWM mode requires top reset value stored here.
  #define SPINDLE_OCRA_TOP_VALUE  0x0100 // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

  // Define spindle output pins.
  #define SPINDLE_PWM_DDR   DDRH
  #define SPINDLE_PWM_PORT  PORTH
  #define SPINDLE_PWM_BIT   5 // MEGA2560 Digital Pin 8
```

Because I'm trying to use that PWM pin. It doesn't work for other pins, either. Enable and direction work correctly.

Thanks for any help you can give!

X3msnake, 2017-09-25:
Are you using a PWM capable pin?

MaxBareiss, 2017-09-25:
I figured it out; the issue was that `SPINDLE_OCRA_REGISTER` needed to be `OCR4A`, not `OCR4C`. I'm going to make a PR soon that makes it clearer that `SPINDLE_OCRA_REGISTER` needs to be the 'A' register for the timer, regardless of which output pin is used.

closed 2017-09-25 by MaxBareiss