gnea/grbl-Mega #40: Feed speed varies?

rodfrey, 2017-09-26:
Hi, I seem to have GRBL controlling my laser alright, but when I try to cut a SVG file (I'm using LaserWeb) the feed seems to really slow down around the arcs and then just take off on the straightaways. 

I've confirmed the gcode only has one feed rate set.

I've done some searching but my google fu is failing me... is this a common thing for GRBL? Am I abusing it somehow? Is there somewhere in the source code I can look to understand what's happening?

chamnit, 2017-09-26:
@rodfrey: There can be several reasons for this. Can you share your $$ Grbl settings, your $I build info strings, your gcode program (as .txt and drag and dropped here), and what GUI you are using.

rodfrey, 2017-09-26:
Thanks for the incredibly speedy response. 

$$ settings:
```$0=10
$1=25
$2=0
$3=3
$4=0
$5=0
$6=0
$10=2
$11=0.010
$12=0.020
$13=0
$20=0
$21=0
$22=1
$23=0
$24=25.000
$25=3000.000
$26=250
$27=1.000
$30=1000
$31=0
$32=1
$100=208.333
$101=208.333
$102=800.000
$110=2000.000
$111=2000.000
$112=500.000
$120=200.000
$121=200.000
$122=10.000
$130=1200.000
$131=900.000
$132=200.000

$I
[VER:1.1f.20170802:]
[OPT:VNM,35,255]
ok
```
[sample_gcode.txt](https://github.com/gnea/grbl-Mega/files/1333700/sample_gcode.txt)

chamnit, 2017-09-26:
Your $$ Grbl settings looks fine, but I would set your $1=255 to ensure your steppers are always enabled. This will help make laser job run smoother if the buffer gets starved. You can use the $SLP command to manually disable the steppers.

The main issues are the length of the line segments in the arcs and the feed rate. They are 0.05mm in length and running at 1500 mm/min. 

The longer you make the lines, the faster Grbl will be able to run through them. The shorter, Grbl may struggle to process the gcode fast enough. This is a function of a lot of things, like GUI streaming speed, Grbl settings, buffer size, processing speed etc. LaserWeb should be ok. Increasing acceleration settings can also help. In general, 0.1 mm per line segment is about the minimum for reasonable performance. 0.2mm for faster jobs. Any less will require feed rate reductions.

At 1500 mm/min, you are right around the max lines per second Grbl can execute on the AVR processor, which is about 400-500 per second. Slow down the feed or use a faster chip, like the ARM Cortex M3 LPC port (~1500 lines per second) or wait for an official Grbl ARM release, which will address some of these performance limitations internally.

closed 2017-09-26 by chamnit