gnea/grbl-Mega #45: Feature request: "Axis auto-squaring"

chaotix-, 2017-11-03:
I'm using a MPCNC with GRBL. I don't know if this happens with other machines as well, but at least with the MPCNC the axis can "twist" a bit. One way to counter this is to make a distance piece and start X and Y axis at one end with the distance piece to the edge so that both axis are perpendicular to the edge.

But for each axis that has one motor per side there is another solution: Install two limit switches and while homing drive both motors in sync as usual but stop stepping the first motor as soon as it hits its limit switch but continue with the second motor until it also hits its limit switch (maybe stop and enter an alarm state if this goes on for more than a centimeter or two). This needs two additional pins per axis: Two limit inputs and two step outputs (the direction pin of the two motors can be shared). If the limit switches are correctly installed the axis should be perfectly straight after that. I think this would only be needed for the X and Y axis as I have not seen a machine yet that has two motors for the Z axis. The MPCNC is probably the most prone to getting the axis skewed, but maybe some other machines can profit from this as well if only to a lesser degree.

The Mega should have enough pins to spare for a feature like this. Port PA (where the step ins are located now) has 8 pins so all additional pins could be on the same port and there is even room to grow to six axis if this should ever be viable in the future.

109JB, 2017-11-27:
One method to easily square a 2 motor driven axis, like a moving gantry machine, is to simply run the axis into a pair of hard stops at slow speed. One side will hit its stop first and that motor will miss steps until the other side hits its hard stop. At that point the axis is squared at least as well as the hard stops are aligned. I don't have a 2 motor axis on my machine, but slowly running into a hard stop is how I used to home my 1-motor axis before I installed homing switches.  Worked within about 0.002".

neomadic, 2018-05-26:
I think this is a good idea but probably not a huge priority.  I've found that manually backing the gantry into two spacer bars with the steppers disabled like you mentioned will line things up really easily, and more accurately than switches can.

zurpv1, 2019-02-06:
Has there been any advancements with this request? Or perhaps any instructions?

Thanks.

lulu3d, 2019-02-11:
Yep, would like to see this possibility too!