gnea/grbl-Mega #46: change USART

WD24, 2017-11-14:
Is it possible to use a different USART on the Arduino mega?

I tried to change the USART in cpu_map.h from:
  #define SERIAL_RX USART0_RX_vect
  #define SERIAL_UDRE USART0_UDRE_vect

to 

  #define SERIAL_RX USART1_RX_vect
  #define SERIAL_UDRE USART1_UDRE_vect

and then connect FTDI232 serial programmer but it out puts nothing. 
Basically the Serial chip on my arduino mega is broken.
is it possible to change to a differnet serial port?

chamnit, 2017-11-14:
Yes. You’ll need to update the macros in serial.c as well. Grbl isn’t fully configurable in the cpu map file.

Diamus, 2018-02-08:
I am trying to get a makeblock megapi to use USART3 in order connect a bluetooth module to the wireless port on the megapi.  I have tried to change the serial.c to reference all the registers related to USART3 but still cannot communicate.  I have reached the end of my skills after several hours of research and trying.  I don't like it but I am at a loss and asking for help.

I have attached the serial.c and cpu_map.h files:

[cpu_map.txt](https://github.com/gnea/grbl-Mega/files/1709253/cpu_map.txt)
[serial.txt](https://github.com/gnea/grbl-Mega/files/1709254/serial.txt)

Diamus, 2018-02-13:
Found the problem.  I missed changing one register.  Corrected it and all is well.

closed 2018-02-14 by chamnit

WD24, 2018-02-23:
Which register did you miss out?

Diamus, 2018-02-24:
Line 103 of serial.c
was
UCSR0B |=  (1 << UDRIE3);
should be
UCSR3B |=  (1 << UDRIE3);

Attached is the fully corrected file.
[serial.c.zip](https://github.com/gnea/grbl-Mega/files/1753648/serial.c.zip)