gnea/grbl-Mega #47: laser connected to digital pin 10

renebohne, 2017-11-26:
Hi,

first, thanks for this awesome project!
I am trying to use it with the Makeblock Laserbot and their MegaPi board.
I mapped all the pins, but my last problem is the laser pin. The MegaPi uses digital Pin 10 (PB4/OC2A).
This means, that I need to use timer2 for PWM on this pin and this is an 8-bit timer... the current implementation uses 16bit-timers.

Can somebody help me with using PB4 for the laser?
This is my current code in cpu_map.h
`
// Advanced Configuration Below You should not need to touch these variables
// Set Timer up to use TIMER2A which is attached to Digital Pin 10 - PB4
// warning: Timer2 is an 8-bit timer!
#define SPINDLE_PWM_MAX_VALUE     255.0 // Translates to about 1.9 kHz PWM frequency at 1/8 prescaler
#ifndef SPINDLE_PWM_MIN_VALUE
#define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
#endif
#define SPINDLE_PWM_OFF_VALUE     0
#define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)

//rb MegaPi THIS HAS TO BE CHANGED TO A DIFFERENT PIN! OC2A is for D10 (PB4)
//Control Digital Pin 6 which is Servo 2 signal pin on Ramps 1.4 board
//#define SPINDLE_TCCRA_REGISTER    TCCR4A
//#define SPINDLE_TCCRB_REGISTER    TCCR4B
//#define SPINDLE_OCR_REGISTER      OCR4C
//#define SPINDLE_COMB_BIT          COM4C1
#define SPINDLE_TCCRA_REGISTER    TCCR2A
#define SPINDLE_TCCRB_REGISTER    TCCR2B
#define SPINDLE_OCR_REGISTER      OCR2A
#define SPINDLE_COMB_BIT          COM2A0

// 1/8 Prescaler, Fast PWM mode
#define SPINDLE_TCCRA_INIT_MASK ((1<<WGM20) | (1<<WGM21))
#define SPINDLE_TCCRB_INIT_MASK ((1<<WGM22) | (1<<CS21))
#define SPINDLE_OCRA_REGISTER   OCR2A // Fast PWM mode requires top reset value stored here.
#define SPINDLE_OCRA_TOP_VALUE  0xFF // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

// Define spindle output pins.
#define SPINDLE_PWM_DDR   DDRB
#define SPINDLE_PWM_PORT  PORTB
#define SPINDLE_PWM_BIT   4 // MEGA2560 Digital Pin 10 - 12v output MegaPi

// Define flood and mist coolant enable output pins.
#define SPINDLE_ENABLE_DDR   DDRF
#define SPINDLE_ENABLE_PORT  PORTF
#define SPINDLE_ENABLE_BIT   0 // MEGA2560 Analog Pin 0 - 12v output
#define SPINDLE_DIRECTION_DDR    DDRG
#define SPINDLE_DIRECTION_PORT   PORTG
#define SPINDLE_DIRECTION_BIT    2// MEGA2560 Digital Pin 39 - MegaPi P15 5x2 pin1

// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_DDR   DDRA
#define COOLANT_FLOOD_PORT  PORTA
#define COOLANT_FLOOD_BIT   2 // MEGA2560 Digital Pin 24
#define COOLANT_MIST_DDR    DDRA
#define COOLANT_MIST_PORT   PORTA
#define COOLANT_MIST_BIT    0 // MEGA2560 Digital Pin 22
`

renebohne, 2017-12-12:
I fixed it. The MegaPi has a super slow mosfet, so I had to decrease the PWM frequency. If somebody is interested, you can have a look at https://github.com/renebohne/grbl-laserbot
I close the issue in this repo, since this was a very special feature request.

closed 2017-12-12 by renebohne

Fusseldieb, 2020-08-18:
@renebohne Since you've already did things with PORTs and stuff, can you help me real quick on how my definitions should look. I'm setting up a Laser and am absolutely clueless on how to make the pin D11 work on the Mega2560

https://github.com/fra589/grbl-Mega-5X/issues/138

renebohne, 2020-08-19:
> @renebohne Since you've already did things with PORTs and stuff, can you help me real quick on how my definitions should look. I'm setting up a Laser and am absolutely clueless on how to make the pin D11 work on the Mega2560
> 
> [fra589#138](https://github.com/fra589/grbl-Mega-5X/issues/138)

You can search for "//rb THIS CHANGED TO A DIFFERENT PIN! OC2A is for D10 (PB4) - using 8-bit timer Timer2"
in file https://github.com/renebohne/grbl-laserbot/blob/master/Arduino/cpu_map.h
And have a general look into my changes in that fork... 
but my recommendation is not to waste time on this and use official hardware instead.