gnea/grbl-Mega #48: Feature proposal : 5 axis for grbl-Mega

fra589, 2018-01-02:
Hi all,

The first version of GRBL-Mega-5X is here : https://github.com/fra589/grbl-Mega-5X
----------------------------------------------------------------------
5-axis version for GRBL Mega with RAMPS 1.4
Since the RAMPS board has 5 stepper driver slots, I added the management of the 2 unmanaged stepper driver slots in the original version of gnea/grbl-Mega.
----------------------------------------------------------------------
The number of axes is parameterized using the variable N_AXIS in the file grbl/nutbolts.h, the accepted values are 3, 4 or 5.
Great care has been taken to maintain compatibility with the original version gnea/grbl-Mega: By setting N_AXIS to 3, we get the same executable as the original GRBL with same other options.
This 5-axis version is currently limited to the use of the RAMP card. The 2 additional axes are the rotary axes A and B.
----------------------------------------------------------------------
TODO: (without order priority, but I will continue to make it better)
- Make the same 5 axis functionalities for other hardware (DEFAULTS_GENERIC + CPU_MAP_2560_INITIAL), perhaps add 6th axis. :-)
- Possibility to choice which axes are enabled for 4, 5 (or 6) : A, B or C rotational axis or U, V, W linear axis.
- ?

If you are interested I can make a push request to merge it in this branch...

Regards, Gauthier.

alfredanil, 2018-02-14:
@fra589 
Would it be possible to add axis cloning to the TODO list for dual Y or else. If it is already implemented please ignore this request. Wonder if Mega has the horse power to run 6 axis, have you thought about porting this to 32 bit MCU like Due (RADDS or Ramps-FD shield) or M4 or M7. Adding S-shape velocity ramp would be even cooler much like G2Core. Maybe even run a web server/web socket as host. Nice job BTW.

fra589, 2018-02-14:
@alfredanil:
Yes I think it will be possible to add axis cloning without too hard job... => Added today to my TODO list :-)
I have no planing for another CPU port.
I tested 5 axes with full 5 axis simultaneous control without lag... 
The architecture of Grbl is very close to the hardware, which makes it very difficult to port to other processors but makes it very powerful. The Mega card largely supports the 5 axes and should not shudder with 6.
In fact, the work of calculating the positioning of the axes is done upstream of the movement during the buffering of the Gcode commands. This consists in calculating the increments of movement on each axis according to the total displacements, the speed and the acceleration of each axis. The whole is then combined into one byte containing one motion bit per axis, another with one direction bit per axis, and so on.
Then, the movements are realized using timed interrupts that are exactly the same regardless of the number of axes driven, since they use only 1 byte per function regardless of the number of axes in motion. That should allow to have the same performances until 8 axes ... But there, I think that there will not be enough ports available at the physical level to do everything :-)

@chamnit:
 By the way, the firsh post of the 5 axis feature proposal make just a sentence and not a question, so you are right to not reply to it :-) 
Now, there is my question :
Are you interested by a push request of my 5 axis for grbl-Mega version ? or do you prefer that I stay and continue alone with my branch ?

Regards, Gauthier.

chamnit, 2018-02-14:
Just as an FYI, Grbl is undergoing an complete overhaul that coincides with the ARM version of Grbl. This is to make it easier to port to various MCUs and make it more maintainable in the future. 

As for the 5-axis stuff, I haven't had the time to really look at it. You're not the first person to post a pull request for this. For now, you can continue with your branch. I'm investing all of my time into the next major version, which will leave the old Grbl code behind in a lot of ways.

closed 2018-02-14 by chamnit

creatron, 2018-08-15:
I ported the software to run on 'CPU_MAP_2560_INITIAL' by just adding some macro's, it seem to operate fine., however I am new to CNC, and  have a four axis available. G1 X10 Y10 Z-10 A10 F120 seem to drive stepper at the same time, indication things seem to operate fine. However I added end stops to my CNC, and do not know how to handle the rotating axis. If I link the mega pin to ground , and send a home command, the rotation axis it just turns and after a while ends with an error. How does one attach a position indicator as it is not a 'real' end stop? . If I move the 'homing' I can home the X, Y and Z successfully. 

While I am asking is moving more than +-360 an error or not?, some web pages seem to say just that , but it does not make sense to me.

Thanks for your enormous contribution..

rodrigorrm, 2018-09-04:
Hello there, 

Which G-code sender program could I use for controlling the 5-axis GRBL?

fra589, 2018-09-05:
Hi @rodrigorrm,  
You can try my cn5X++ (https://github.com/fra589/cn5X).  
I am writing this new 5/6 axis Grbl control panel to implement all the grbl-Mega-5X capabilities because I don't find any good software to do it.  
Then, you will be new beta tester.  
@++;  
Gauthier.

fra589, 2018-09-06:
Hi,
I mean full 5 axis control. The last version is also able to make 6 axis full control ! :-)  
It depend of my last version of Grbl-Mega-5X wich have an output functionality wich say his axis number and there names. So, it can manage both full axis independantly or cloned axis like dual Z axis.  
  
@++;
Gauthier.

geargh, 2018-09-25:
Hi, 
After Compile and installed, EEPROM value wrong($$).
Original gnea version was same.

fra589, 2019-02-05:
Hello @antoniogt ,  
[grbl-Mega-5X](https://github.com/fra589/grbl-Mega-5X) have a real capability of 6 axis, and the port of capability to up to 8 axis will be arrive soon.  
Since I didn't find any Gcode sender able to implement all the 5X capability, I decided to create my own sender : [cn5X++](https://github.com/fra589/cn5X). It's still an alpha version, but it begin to be usable for real milling.
The GCode generation directly from stl is too complex to be fully auto. You need CFAO softwares like Fusion360, or better, the open source [FreeCAD](https://www.freecadweb.org/) with his great module [Path](https://www.freecadweb.org/wiki/Path_Workbench). It still under active developpment, but realy promising.
@++;
[Gauthier](https://github.com/fra589)

antoniogt, 2019-02-05:
It sounds great! Im very glad to read that there is some options to run with 5 axis, ill download your cn5x++ but it only aviable with Linux or with windows or Mac OS as well? Im using Mac.
I was taking a look about Freecad Path but I only found tutorial for 3 axis, do you know if Freecad Path can manage 5 axis too.
And what about if i use Fusion360? How must i wire the arduino mega 250 + Ramps 1.4 with the GRBL 5 mega flashed inside? Just by the USB connector? 
Thank you very much by your fast response

dodidada, 2019-02-05:
hi, actually am so interested a ur grbl 5 axis project and its so needed ,but after downloading the grbl 5x it gave me some errors while verifying , why?!, sorry am still a newbie ;), i opened marlin, downloaded the sketch , verifying then errors like "no such files or dictionary" any advice will be appreciated ,thanks

dodidada, 2019-02-06:
@fra589, you are the best!, realy man u cant imagine how much time it took for me to find such configuration for mega with ramps 1.4, i want to thank you so so much for that great work you did, i tried the 5x file , well it seems working good, as am new in cnc world in general and still learning , i faced some issues in ugs like error 9 and error 20 , but am sure with some learning i could solve it, i really cant wait to try your cn5x, downloaded , not tested yet, again , thank u so much and keep it up.

fra589, 2019-02-06:
Hi @antoniogt,  
I think cn5X++ should work on Mac if requisits can be installed on it (Python3, PyQT5 and PyQT5-QtSerialPort (automatly installed on Windows when installing pyqt5 with pip command)).
I don't have a Mac, so, I can't be sure.
Actually, FreeCAD have capability of 3 axis under his stable version, but under very active developpment, it allready capable of 4 axis : https://www.youtube.com/watch?v=6sEzrnasdgo
I think the 5 axis can come not too late...
Fusion 360 have 5 axis capability, but I think only in his ultimate version... Not free !
To flash grbl-Mega-5X on Arduino Mega, you only need USB connection and the ad-hoc software, for example, Arduino software work weel for this : https://github.com/grbl/grbl/wiki/Flashing-Grbl-to-an-Arduino
@++;
Gauthier

dodidada, 2019-02-06:
hey @fra589 , could you pls help me with that issue?, when i run ramps board and connecting the usb, all 5 stepper motors freeze immediately!, as i mentioned , am new in all of this but as i learned a bit that something with wrong stepping parameters could do this, am not sure , i passed error 9 and now stuck in error22!, now when i plug ramps with electricity sure nothing happen, but time usb connected , even without start any command or any programs, all stepper freeze, any advice will help, thanks in advance.

dodidada, 2019-02-07:
maybe that could help..$$ result
Grbl 1.1i {'$' for help}
{MSG:'$H'| '$X' to unlock}
$0=10
$1=255
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.020
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=500.000
$25=2000.000
$26=250
$27=5.000
$30=0
$31=0
$32=0
$100=80.000
$101=80.000
$102=4000.000
$103=8.889
$104=8.889
$110=9000.000
$111=9000.000
$112=300.000
$113=1440.000
$120=300.000
$121=300.000
$122=100.000
$123=100.000
$123=100.000
$124=100.000
$130=200.000
$131=200.000
$132=200.000
$133=360.000
$134=180.000
ok

antoniogt, 2019-02-07:
Hello, i just ordered an Arduino 250 + ramps 1.4 board kit in order to put my hands in this project and try to help as well, ill try to load the Arduino sketch and connect the motors of my 3d printed and asses the beheavour.
Anyway i just saw that there is an option to use 5 axis through g-code generation from Blender, its named “Blender Cam” but its quite hard difficult to install but it runs with 4 and 5 axis for free but there is not su much support with tutorials in internet

dodidada, 2019-02-07:
i should learn some coding first, try to fix problems and then come back with questions!, thanks for your support guys

dodidada, 2019-02-08:
hey, its finally worked good!, after changing some $$ settings as am using nema 17 1.8 degree with jumpers they now moving way smother,  am using cncjs controler i tried only 4 axis (with A). no 5th axis, and they all works wonderfully good, i tried to use cn5x++, i download python 3.7 and........,ok i dnt know how to open the file :-l

dodidada, 2019-02-08:
oops , forgot to open QSerialPort!

BernardG, 2019-05-27:
> Fusion 360 have 5 axis capability, but I think only in his ultimate version... Not free !

Nope! You have ALL Manufacture capabilities (new name for the CAM module) in the free licences (and not versions). Just in case you don't realize this, Fusion 360 CAM is the same module as the HSM module in Inventor, and in HSMWorks, module for Solidworks....

The limit is your imagination, and abilities, not the software....

antoniogt, 2019-05-27:
So i supose that once that you design your milling strategy in Fusion 360 Cam you need a software to send the g-code to the brakeboard like Mastercam or Mach3?
The thing is what about the driver for nema 17 control, or if you want to use nema 23 with bigger drivers, how do you wire it? And another question is about all the wiring in the board between components, we dont have it? And Ramps has actually 5 outputs for stepper drivers (3 XYZ axis and 2 extruders) how do you change it for use them as individual axis? in the firmware i mean
Thanks again for all your effort

clement666946, 2019-07-15:
boujour a tous. 
salut   @fra589. 

je fait une rapide présentation. 
depuis peut de temps dans les programme de domotique et cnc.
en cour de conception d'une imprimant 3d et en attente d'en recevoir une.

technicien outilleur, bts en industrialisation des produit mécanique. je suis tourneur fraiseur de métier. d'ou l'utilisation avancer dans le g code.

je vien juste de voir cette discussion. est je suis extrêmement interaisser par ce que tu est en train de faire. 
vue que j'ai aussi l'intention de concevoir une cnc 5 axe minimum vue que je dispose de logicielle pour concevoir du g code multie axe meme si je serais capable de l'ecrire sur papier ce qui serait inutile.

bref. je ne savait pas comment vous joindre d'ou le fait du message sur le forum. 

comment peut on vous joindre ci l'on ce trouve en difficulté via votre version de grbl ?

vue que je ne suis pas un grand specialiste en programation d'arduino et encore moins dans la domotique.

sur ce merci pour votre travaille en espérant pouvoir ce tenir au courant. cordialement.

je vais tester votre fichier d'ici peut de temps ;)

fra589, 2019-07-16:
Bonjour @clement666946,  

Ma version grbl-Mega-5X est hébergée ici : https://github.com/fra589/grbl-Mega-5X  
Les questions et bugs rencontrés peuvent y être discutés dans l'onglet "[Issues](https://github.com/fra589/grbl-Mega-5X/issues)".  
Ne pas hésiter à chercher dans les questions déjà posées et résolues (https://github.com/fra589/grbl-Mega-5X/issues?q=is%3Aissue+is%3Aclosed) avant d'en poser une nouvelle... C'est sûr, l'usage de l'anglais rend la compréhension plus difficile, mais Google translate marche suffisament bien pour m'éviter de répéter trop souvent les mêmes choses. :smile:  

@++;  
Gauthier.

interweb-md, 2020-02-11:
Hi Gauthier,
I've done 4 axis rewrite for grbl few years ago and machine is working fine so far. It was needed for hot wire machine control, there are 2 towers with 2 axis each, so i needed 4  independent axis. The thing is how to limit feed rate. Now i need to upgrade this machine and add nother 5th axis. I've found your 5 axis firmware but here is my issue. Your firmware is basicaly 3 axis machine + 2 independent axis. It means feed rate is calculated based on tool motion in space held by 3 axis. It's wrong in my case.  I suggest to rewrite grbl in a universal manner. For example you have 5 axis machine and in configuration file I want to set configuration for example 2 axis + 2 axis + 1 independent axis. In this case feed rate is calculated for first pair, then second so the maximum of these two has to be the actual feedrate. Hope you understand. So my question is if it is possible to refactor grbl in a configurable manner so you can configure groups for different setups.
Thank you.

fra589, 2020-02-11:
Hi @interweb-md,
My 5/6 axis firmware version is not a 3+2 or 3+3 axis, but a full 6 axis version.
The speed is calculated for all axis to have linear movment with all axis ending at the same time with the good speed.
The speed calculation is pretty complicated because it also compute acceleration on all axis to ensure than no max speed or max accel are overshooted.
Try to see  in the file planner.c, where the speed is calculated if you find how to adjust it for your need...
@++;
Gauthier.

ozkankara3329, 2020-04-19:
I make ramps 1.4 cnc. I will use double motor for y and z axes. Double output E0 will be double for Y axis. E1 output will be double for Z axis. I am glad how you can make the settings in gbrl mega 5x.

ozkankara3329, 2020-04-19:
DEAR fra589 I would be glad if you can help ..

fra589, 2020-04-20:
Hi @ozkankara3329,  
1) Read all the docs on the wikis pages : 
- https://github.com/gnea/grbl/wiki
- https://github.com/fra589/grbl-Mega-5X/wiki
2) Read the old issues ([active](https://github.com/fra589/grbl-Mega-5X/issues) and [closed](https://github.com/fra589/grbl-Mega-5X/issues?q=is%3Aissue+is%3Aclosed)), perhaps somebody have alreayd asked similar question,
3) Ask [Google](https://www.google.com/) for your question,
4) Post your question on the right project issues: [grbl-Mega-5X](https://github.com/fra589/grbl-Mega-5X/issues) and not on this one wich is the original grbl-Mega project.

ozkankara3329, 2020-04-20:
*_*I named AXIS 4 Y.
I named AXIS 5 Z, but;
Y is still running from drive A.
Z also works from drive B.
What am I doing wrong.**_

 #define N_AXIS 5            // Number of axes
  #define N_AXIS_LINEAR 5     // Number of linears axis
#else
  #define N_AXIS 3 // Number of axes = 3 if not DEFAULTS_RAMPS_BOARD
#endif

#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X' // Axis names must be in X, Y, Z, A, B, C, U, V & W.
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'

#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  **#define AXIS_4_NAME 'Y' // Letter of axis number 4
#endif
#if N_AXIS > 4
  #define AXIS_5 4
  #define AXIS_5_NAME 'Z' // Letter of axis number 5
#endif

ozkankara3329, 2020-04-20:
Can you send the hex file with the drivers as below.
Hex file compiled with N_AXIS = 5 & axis names = X, Y, Z, Y & Z

fra589, 2020-04-20:
No, I can't.  
  
I don't have the capacity to offer hex files for all possibles configuration and for all users who ask me.  
  
1) read the docs: https://github.com/gnea/grbl/wiki, https://github.com/fra589/grbl-Mega-5X/wiki
2) read the docs: https://github.com/gnea/grbl/wiki, https://github.com/fra589/grbl-Mega-5X/wiki
3) read the doc: https://github.com/gnea/grbl/wiki/Compiling-Grbl
4) adapt grbl to your need and compile it and upload it to your Hardware by yourself
5) ask in the [right issue page](https://github.com/fra589/grbl-Mega-5X/issues) (not this one!!!) if you have a real issue which doesn't have already been replyid, without forget to give all necessary details of your config.
  
Help yourself and heaven will help you.

@++;  
Gauthier.