gnea/grbl-Mega #49: CoreXY homing problem

bajticzek, 2018-01-02:
Hello,
I built a CoreXY laser machine and I am using Mega2560+RAMPS combo with grbl-Mega 1.1f. I have a problem with homing. Simply put after initiating homing, only one motor moves at a time. That means when machine should be homing X first, then Y, it doesn't. The gantry moves diagonally because only one of the motors moves at a time. Jogging works without an issue. I tried Repetier firmware and there homing and jogging works both flawlessly. But it isn't grbl, which I want to use.

I downloaded fresh .zip file, extracted the grbl folder to Arduino\libraries and edited following in config.h (pure minimum to get things working):

- Commented lines 37 and 38 `//#define DEFAULTS_GENERIC`, `//#define CPU_MAP_2560_INITIAL`
- Uncommented lines 41 and 42 `#define DEFAULTS_RAMPS_BOARD`, `#define CPU_MAP_2560_RAMPS_BOARD`
- Commented line 112 `//#define HOMING_CYCLE_2 (1<<Z_AXIS)` and left line 110 and 111 uncommented `#define HOMING_CYCLE_0 (1<<X_AXIS)`, `#define HOMING_CYCLE_1 (1<<Y_AXIS)`
- Uncommented line 190 `#define COREXY`
- Uncommented line 208 `#define INVERT_MIN_LIMIT_PIN_MASK ((1<<X_AXIS) | (1<<Y_AXIS) | (1<<Z_AXIS))` to make my limit switches work as they should

This was my second attempt in case I missed something on the first try. I tried to edit the config.h as little as possible. But still homing doesn't work. Can you guys tell me what am I missing?

Thanks for any info.

bajticzek, 2018-01-03:
After some extensive digging in the source code I solved this issue by altering line 347 in `limits.c`. I changed `sys.homing_axis_lock[idx] = axislock[idx];` to `sys.homing_axis_lock[idx] = 65;` which basically doesn't lock the movement to one motor - 65 is combined mask for step pins on RAMPS board for motor A and motor B. I consider this a bug and my workaround is really not a final solution. Can anyone who is more familiar with grbl source code look into this issue and make a bulletproof fix? 

Thank you.

chamnit, 2018-01-03:
@bajticzek : Thanks for digging and locating the bug. @docwelch Do you have time to look into this?

docwelch, 2018-01-04:
@bajticzek  and @chamnit:
I don't have a CoreXY machine so have not seen this issue. I relied solely on the changes @jekhor had made here so I would defer to him. I will look a little deeper but am not sure I'll be able to quickly determine the cause. As was pointed out when RAMPS support was added, it's a temporary solution and not an ideal one.

bajticzek, 2018-03-14:
I recently added Z-Axis to my Laser machine, so if anyone is interested in homing Z too, change `sys.homing_axis_lock[idx] = 65;` to `sys.homing_axis_lock[idx] = 73;` to allow Z homing.