gnea/grbl-Mega #51: Grbl has not finished booting. (Using 115200 baud)

therobotguyllc, 2018-01-18:
Greetings,
I was able to get my Uno grbl running without any issues on a few different operating systems. However, I am trying to get a Mega loaded up since it's got more hardware resources and it's just not working for me. 

I have tried every solution online it seems with no luck. I have tried every baud rate possible. I have use UGS, bCNC, SourceRabbit GCode Sender, and none of them connect to the Mega after loading grbl. I have used XLoader to upload the HEX file version and I also tried uploading with the Arduino IDE. None of them work. The HEX file was "grbl-Mega_v1.1f.20170802.hex", which appears to be the latest. 

I am out of ideas to get this thing working. I saw a number of guys on youtube make this work with the identical pieces of equipment that I am using so I do not know what the issue is. 

Any suggestions would be greatly appreciated.

chamnit, 2018-01-18:
It sounds like the flashing process failed. You should be able to connect to Grbl via a terminal (Arduino IDE at 115200 baud). Pressing enter should reply with an 'ok'. If not, then you'll need to fix the flashing issue. I would probably just follow the compiling instructions in the Grbl wiki pages.

shooter64738, 2018-01-21:
One other thought. One of my 2560 boards will not flash, if anything is connected to ANY of the serial pins. The other boards only fail if something is connected to serial 0. But that one board wont flash anything if there is anything connected to any of the other serial pins (serial 0 or not). Maybe you don't have anything connected to those other pins, but thought it worth a  mention.

xXSyl20Xx, 2018-02-28:
I had the same problem and found the solution.
I use Arduino IDE to compile and upload.

GRBL files shall be in the "librairies" folder, make sure that no other/older GRBL files already are present.
I had an old version of grbl for Uno board in my "libraries" folder and these files were compiled instead of the files for Mega board.

Cheers
Sylvain

djchikito, 2019-03-03:
> Tuve el mismo problema y encontré la solución. 
> Yo uso Arduino IDE para compilar y subir.
> 
> Los archivos GRBL estarán en la carpeta "librairies", asegúrese de que no haya otros archivos GRBL / anteriores que ya estén presentes. 
> Tenía una versión antigua de grbl para el tablero Uno en mi carpeta de "bibliotecas" y estos archivos se compilaron en lugar de los archivos para el tablero Mega.
> 
> Saludos 
> Sylvain

Buenas, mismo problema, en la carpeta "librerias" tenia la carpeta antigua de GRBL(el que es para Arduino Uno), borre esa carpeta (o se le cambia el nombre), copias la carpeta bajada de aqui y podes subir tranquilo.

Gracias por la respuesta Sylvain!

Saludos

DEKO-TECH, 2020-06-09:
Hola a todos, soy nuevo en la comunidad y me está pasando el mismo error con Universal GCode Sender, pero tengo las librerías correctamente instaladas y me dice conectado correctamente pero cuando le pongo los dos $$ en la barra de comandos, me tira ese error de GRBL HAS NOT FINISHED BOOTING.

¿Alguien puede ayudarme?

Muchas gracias de antemano