gnea/grbl-Mega #54: Compile for windows

JosephusMiller, 2018-03-09:
Hi,

after >1 hour trying to install the GRBL libs through the Arduino IDE I somewhere found a description to simply 
- copy the grbl folder to the libraries,
- restart the ide
- ready for compilation.

Worked instantly. 

I think this should be mentioned on the installation / compile section in the grbl wiki. At least as an alternative. 

Best regards,
JM.

chamnit, 2018-03-09:
The wiki instructions basically do the same thing. There is some extra instructions that are there to prevent common errors, like not updating the IDE or not removing an old Grbl installation.

closed 2018-03-09 by chamnit

Chrominator, 2019-09-11:
The wiki instructions did not made the job for me, IDE complaining about the missing .zip.
The procedure described by JosephusMiller actually works and it avoids to loose more time on compiling grbl-mega.