gnea/grbl-Mega #55: PWM on Ramps with grbl-MEGA not working.  Only 0 to 5 volts nothing in between...

bigshug, 2018-03-31:
Hey folks.  I can’t get PWM to function properly on Sainsmart RAMPS Mega2560 board with grbl-Mega.

First off - I’m checking output on D11.  Correct pin for RAMPS?

It’s all or nothing - 0 or 5 volts.  I can switch it on and off with M3 and M5 but get no change when changing the S value.

I’ve been through the config.h spindle settings and checked cpu_map.h but nothing fixes this issue.

Everything works as it should on an UNO with normal GRBL but it would be useful to get some use out of this board.

Any ideas on how to sort this before this board is consigned to the parts bin?

Cheers

bigshug, 2018-04-01:
Ok.  Reflashed and now get different results - why I do not know...

D11 = is now 0.1v
D6   M4 gives 5v  - M3 turns it off - but M5 doesn’t switch it off - (I guess this is spindle direction?)
D5  M3 and M4 give 5v - M5 turns off.

bigshug, 2018-04-01:
ok -so it looks like PIN 8 is the spindle PWM output.  How do I define another PWM pin for testing purposes?  Will I need to change the register values too?  

Yes I am in way over my head but really want to figure this out now :o)

bigshug, 2018-04-02:
Right then.  Sorted.  Pin 8 is indeed the PWM pin.  However it is 12v when the board is externally powered and I'm after 0 - 5volts.  There is 5 volts going to the mosfet - I guess this is the real D8.  Will trace it back somewhere and take a feed off it.

chamnit, 2018-04-02:
@bigshug : The Arduino Mega does not operate at 12V. If you are reading 12V, you are doing something wrong or something is wrong with your board. It should be 0V to 5V. 

FYI, cpu_map.h contains the pin configuration and shows you that D8 is the PWM pin.

closed 2018-04-02 by chamnit

bigshug, 2018-04-02:
The board is a 'Sainsmart all in one' that can be powered externally - I guess for high power heated bed etc.  It gives a 12v PWM output at the 'BED' terminal or 5v PWM on D8 going into the FET.  It's a nice board so glad it's now working.
[image]

bigshug, 2018-04-02:
See comment here about 12v - https://github.com/gnea/grbl-Mega/issues/11#issuecomment-311184454

rasciodc, 2018-07-08:
**0-5V PWM on Pin D6:**

I know this issue is closed, but, just in case someone else comes accross the same need of having a 0-5V PWM pin to control the Spindle, here is the code to have it working on pin D6 (Servo 2 on RAMPS). This should replace lines 259 to 283 on cpu_map.h:
```
// Advanced Configuration Below You should not need to touch these variables
  // Set Timer up to use TIMER4C which is attached to Digital Pin 6 - Ramps Servo 2
  #define SPINDLE_PWM_MAX_VALUE     255.0 // Since it's a 8bit timer
  #ifndef SPINDLE_PWM_MIN_VALUE
  #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
  #endif
  #define SPINDLE_PWM_OFF_VALUE     0
  #define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)

  #define SPINDLE_TCCRA_REGISTER    TCCR4A
  #define SPINDLE_TCCRB_REGISTER    TCCR4B
  #define SPINDLE_OCR_REGISTER      OCR4A
  #define SPINDLE_COMB_BIT          COM4A1

  #define SPINDLE_TCCRA_INIT_MASK (1<<WGM41)
  #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41)) 
  #define SPINDLE_OCRA_REGISTER   ICR4 
  #define SPINDLE_OCRA_TOP_VALUE  0xFF // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

  // Define spindle output pins.
  #define SPINDLE_PWM_DDR   DDRH
  #define SPINDLE_PWM_PORT  PORTH
  #define SPINDLE_PWM_BIT   3 // MEGA2560 Digital Pin 6
```

Cheers!

luisluna2, 2018-09-02:
Hello friends, I have a question, how make 3 PWM with D8, D9 and D10?

JeffHank1, 2019-11-11:
A Big Thanks to rasciodc for figuring this solution out and Posting it. It was very easy to cut and paste the solution into the appropriate file, recompile in Arduino IDE, create a Hex file and Upload it to my Arduino Mega 2560 with Xloader. Easy. Works perfectly with my 12V laser that wants a 5V TTL control signal.

ksdfpr, 2020-01-13:
> **0-5V PWM on Pin D6:**
>
> I know this issue is closed, but, just in case someone else comes accross the same need of having a 0-5V PWM pin to control the Spindle, here is the code to have it working [...]

Nice! Could you help me with another board? (GT2560) What have I to change to work with any of the bed pins?

thesalcedo, 2021-01-06:
> A Big Thanks to rasciodc for figuring this solution out and Posting it. It was very easy to cut and paste the solution into the appropriate file, recompile in Arduino IDE, create a Hex file and [...]

@JeffHank1 for this to work, does the laser 12V power need to come from the RAMPS +12V output (D8)? Since the - is shared? I was planning on powering my Laser controller externally.

fedex000000000001, 2023-01-02:
> **0-5V PWM en pin D6:**
> 
> Sé que este problema está cerrado, pero, en caso de que alguien más tenga la misma necesidad de tener un pin PWM de 0-5V para controlar el eje, aquí está el código para que funcione en el pin D6 (Servo 2 en RAMPS). Esto debería reemplazar las líneas 259 a 283 en cpu_map.h:
> 
> ```
> // Advanced Configuration Below You should not need to touch these variables
>   // Set Timer up to use TIMER4C which is attached to Digital Pin 6 - Ramps Servo 2
>   #define SPINDLE_PWM_MAX_VALUE     255.0 // Since it's a 8bit timer
>   #ifndef SPINDLE_PWM_MIN_VALUE
>   #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
>   #endif
>   #define SPINDLE_PWM_OFF_VALUE     0
>   #define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)
> 
>   #define SPINDLE_TCCRA_REGISTER    TCCR4A
>   #define SPINDLE_TCCRB_REGISTER    TCCR4B
>   #define SPINDLE_OCR_REGISTER      OCR4A
>   #define SPINDLE_COMB_BIT          COM4A1
> 
>   #define SPINDLE_TCCRA_INIT_MASK (1<<WGM41)
>   #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41)) 
>   #define SPINDLE_OCRA_REGISTER   ICR4 
>   #define SPINDLE_OCRA_TOP_VALUE  0xFF // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.
> 
>   // Define spindle output pins.
>   #define SPINDLE_PWM_DDR   DDRH
>   #define SPINDLE_PWM_PORT  PORTH
>   #define SPINDLE_PWM_BIT   3 // MEGA2560 Digital Pin 6
> ```
> 
> ¡Salud!

hello, do i have to enable $32 = 1 in grbl to use the PWN of pin d6 ?

luisluna2, 2023-01-02:
Te agradezco mucho tu apoyo y atención, lo estaré probando en la semana, muchísimas gracias.

¡Feliz inicio de Año 2023!