gnea/grbl-Mega #57: Parser state message question

barma1ey, 2018-04-03:
As far as I know, grbl doesn't allow mixing $-commands ($G in this case) and regular G-codes together. 
Even if it does, the solution is ugly. Is there a simple way to keep active codes up-to-date while streaming? Wiki says nothing about this. Thanks.

winder, 2018-04-03:
Most grbl interfaces end up implementing a gcode parser eventually and keep track of the state in parallel to grbl. Here are a few examples:
* UGS: [GcodeParser.java](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/gcode/GcodeParser.java#L377)
* bCNC: [CNC.py](https://github.com/vlachoudis/bCNC/blob/2fd88d24b16c9b27e403941f19d78d3a398191cf/CNC.py#L1211)
* Grbl-Panel: [GrblState.vb](https://github.com/gerritv/Grbl-Panel/blob/a2f66b7ed720c72165bccfc9288f080027300c48/Grbl-Panel/GrblState.vb#L92)

chamnit, 2018-04-03:
@barma1ey : It's not clear why you would want to combine Grbl $ commands with a line of g-code. Unless you want to know the parser state after each g-code block. If so, that is a waste of bandwidth. It's usually not necessary to know the state that closely. Do you have a particular application that requires it?

barma1ey, 2018-04-03:
@winder Yes, i know this. Of course, the most obvious way is to parse each line before sending it. But the planner says it is not so good idea. Need to implement a queue

@chamnit  I am planning to install some kind of tool change device on my machine. semi-automatic maybe. The idea was to pause program (M1  M6 Tx) when change is required, and state string was intended to notify operator which tool to use. Other codes would be useful, too

chamnit, 2018-04-03:
@barma1ey : In that case, it would be best to analyze the g-code prior to sending tool change g-codes (not M6 itself since Grbl doesn't support it). You can easily wait until you get a reply from Grbl upon parsing the prior g-code line before doing anything related to a tool change. No need to constantly query for g-code parser state.

closed 2018-04-03 by chamnit