gnea/grbl-Mega #59: GRBL Mega - Axis Support Naming

JorgenCollinson, 2018-04-09:
hi,

Am looking to build a machine with additional Axis's
Mainly Dual Z Axis and an extra Lathe(Rotary) Axis.

I understand that the Rotary Axis is called Axis 'A'
Can i get input for what you would call a second 'Z' or 'Z2' Axis if you have a lathe axis as well..

fra589, 2018-04-09:
In Gcode, axis are only named with 1 letter. So you can't use Z2 for naming an axis, as 2 is the deplacement length for the Z axis when you use Z2...
The naming standard for additionals axis is A for rotational with rotation axis paralele to the X axis, B for rotational with rotation axis paralele to the Y and C for rotational with rotation axis paralel to the Z axis. For linears additionals axis, the naming convention is U for additional axis paralele to X, V for additional axis paralele to Y ans W for additional axis paralele to Z. 
See http://linuxcnc.org/docs/html/gcode/machining-center.html for more details.
With Grbl, you will need to edit sources yourself to change the axis names.
If you need more than 3 axis, you will need to use another fork. For example, my 5 axis fork in https://github.com/fra589/grbl-Mega-5X

JorgenCollinson, 2018-04-09:
Thanks for the feedback, greatly appreciated.
I will check out your firmware.

Regards
Jorgen

From: Gauthier Brière
Sent: Tuesday, 10 April 2018 2:59 AM
To: gnea/grbl-Mega
Cc: JorgenCollinson; Author
Subject: Re: [gnea/grbl-Mega] GRBL Mega - Axis Support Naming (#59)

In Gcode, axis are only named with 1 letter. So you can't use Z2 for naming an axis, as 2 is the deplacement length for the Z axis when you use Z2...
The naming standard for additionals axis is A for rotational with rotation axis paralele to the X axis, B for rotational with rotation axis paralele to the Y and C for rotational with rotation axis paralel to the Z axis. For linears additionals axis, the naming convention is U for additional axis paralele to X, V for additional axis paralele to Y ans W for additional axis paralele to Z.
See http://linuxcnc.org/docs/html/gcode/machining-center.html for more details.
With Grbl, you will need to edit sources yourself to change the axis names.
If you need more than 3 axis, you will need to use another fork. For example, my 5 axis fork in https://github.com/fra589/grbl-Mega-5X

cprezzi, 2019-02-19:
Grbl doen't support dual axes, but you could just connect two stepper drivers to the same step & direction pins.