gnea/grbl-Mega #64: Homing and realtime status reporting

barma1ey, 2018-05-29:
grbl stops reporting anything while homing. Not a single byte. I have tested on three different senders, the result is the same - nothing happens, all programs (mine too) happily keep stating that machine is idle when IT IS NOT. It should send 'Home' and presumably update its positions shouldn't it?

chamnit, 2018-05-29:
Yes it should say something while homing, but it doesn't. This was an oversight on my part on implementing some kind of feedback mechanism during homing. I can't change the communication protocol until the next version. You'll have to wait until then to see this fixed.

closed 2018-05-29 by chamnit

109JB, 2018-05-29:
You can easily have your sender report that a homing cycle is active.  I have mine set to do this and it is rather easy. You don't need Grbl to tell you it is homing because presumably your sender just sent a $H command. When it does this, I simply change the value of the machine status variable to "Homing".  Since there are no status reports coming from Grbl, this machine status variable won't get changed until homing completes or fails and then will get set to whatever it really is by the status report parsing routine.

barma1ey, 2018-05-29:
Obviously the workaround can make a trick. I thought about that. but I'm still hoping the long awaited update is just around the corner and this soon will be fixed along with G4 issue

barma1ey, 2018-05-30:
Hm. g4 also sends IDLE when definitely it shouldn't . the issue was discussed several times. [https://github.com/gnea/grbl/issues/66](url) for instance

chamnit, 2018-05-30:
@barma1ey : Ah yes. Thanks for jogging my memory. Yes this is still an open issue and again won't be fixed until the next version due to it altering the communication protocol. It's worse to deal with a bunch of broken GUIs.