gnea/grbl-Mega #69: Spindle & Laser on one machine

bgort, 2018-08-10:
Is there any interest in a PR with modifications that allow a spindle and laser to be connected at the same time using different PWM pins, different PWM frequencies, etc. ?  I have my laser mounted next to my spindle and have working modifications to grbl-Mega (RAMPS) to use both - spindle connected to D8/OC4C and laser connected to D7/OC4B - with the active 'head' determined by the $32 (BITFLAG_LASER_MODE) setting (though I still have a safety switch (key) to power the laser, of course).

If so, I can clean up what I've done and put together a PR.

chamnit, 2018-08-10:
This would be nice.

bgort, 2018-08-10:
Will try to clean it up and make a PR soon.

A quick question for you:  I'm not sure if this is a bug or feature, but I've noticed that when in laser mode, the laser doesn't come on with M3 until after a G1 move has been executed, then it stays on.  G0 will turn it back off and it then stays off.  M4 behaves as I thought it would - the laser is only on during G1 moves, off when stopped and during G0 travels.

Would it be better if M3 turned the laser on regardless of whether the carriage is in motion or has moved - for focusing, etc. - while M4 would only turn it on for G1 moves, and off for stopped/G0 travels, etc.?  Let me know what you think.

chamnit, 2018-08-10:
@bgort : M3 staying on was a request from the laser web community, as a way to help focus the machine. All this is written up on the wiki page.

bgort, 2018-08-10:
I think it's a good idea - I think with M3 it should stay on for focusing, etc. - but my point was that when in laser mode, with M3 it doesn't turn on until there's a G1 move, and then it turns back off with a G0 move.  I was asking if it'd be better to simply turn it on and off with M3/M5, like normal, rather than tie on/off to G1/G0 like with M4.

chamnit, 2018-08-10:
@bgort : Sorry. Not enough coffee today. Not sure what you'd gain from allowing G0 moves with M3 to keep the laser on. It's simpler to just state that no laser power will occur when G0 is enabled. Laser will be powered only when G1/2/3 are active. M4 is the only outlier, where it will only activate during motion.

bgort, 2018-08-10:
No worries.  I don't need to actually move it with the laser on, it's just - to me, and maybe I'm weird - counter-intuitive that after homing, jog positioning with G0, etc. - that I'd then need to move with G1 before I could turn the laser on with M3 to focus on the material.

Right now what I have to do, from startup, is:
 1) Home
 2) Position to material location with G0 jogging
 3) Setup zeros, workspace, etc.
 4) $32=1 (to activate laser mode, which in mine selects the laser pin instead of the spindle pin)
 5) M3 S300 (focus power, my max RPM is 21000) - here the laser doesn't actually turn on yet
 6) ***G1X0.1F100 - here the laser turns on***
 7) Focus, etc.
 8) Carry on as normal.. run gcode, etc.

What I'm suggesting is that instead of requiring the extra G1 step (6), if M3 should just turn the laser on and off without regard for G0/G1.

I can and probably will make the necessary changes in the version I'm using.  I'm mostly just asking if this makes sense to provide as a PR, as it seems like it would be better than requiring the G1 before the laser powers on (after M3) while in laser mode, generally.

(Sorry if that's confusing - feels a bit hard to explain properly.)

chamnit, 2018-08-10:
So, it's more of an issue with allowing the laser while jogging?

bgort, 2018-08-10:
No, it's a matter of allowing the laser to be powered on without having to use G1 (or G2, G3) first.  It's an extra manual step.

chamnit, 2018-08-10:
The point of this is to make it more difficult to turn on the laser than turn it off. The average user will have to actively know what they are doing to get things to work. That's arguably overall more safe. There has already been a few discussions on this when the laser mode was first installed. You can read them to see why this was decided.

You can also turn on the laser in one command. G1 can be programmed without a target position. 
'M3 S300 G1 F100'. That will turn on the laser without a motion involved.

bgort, 2018-08-10:
Ah, thanks.  As I mentioned, I wasn't sure if it was a feature (it's clearly 'safer') or a bug.  

For me, it was an issue because I'd like to be able to turn the laser on with the spindle button in bCNC (or whatever other interface - not a fan of bCNC) without having to manually G1, but I can just create/assign a macro.

And thanks again:  I didn't realize G1 could be used without specifying an axis/target.

Appreciate your time/replies.