gnea/grbl-Mega #70: How to change Spindle PWM Frequency

timdavies91, 2018-08-26:
Hi, just want to double check that if I want 8KHz I'd change:

#define SPINDLE_PWM_MAX_VALUE     1024.0 //1.9 KHz
to 
#define SPINDLE_PWM_MAX_VALUE     4096.0 //~8 KHz?

Thanks,
Tim

timdavies91, 2018-08-26:
Looks like it needs to be lower for faster frequency, so guessing set to 255 for ~8 KHz and then update the hex value of SPINDLE_OCRA_TOP_VALUE to 0x00FF?

chamnit, 2018-08-27:
@timdavies91 : Changing frequency has to do with the timing prescaler. There should be a comment somewhere in the cpu_map.h code to show you what to alter.

closed 2018-08-27 by chamnit

timdavies91, 2018-08-27:
@chamnit thanks for the reply! I can see the comments in the standard 328p version of GRBL but looks like this code base has been modified to work with the Mega and those comments have been removed unfortunately. Might have to do some more reading and figure out which bits I need to change for the prescaler, this would be  a lot easier with a scope...

chamnit, 2018-08-27:
@timdavies91 : Sorry, forgot that the Mega was different and using a 16-bit timer for the PWM. So I think you have to change both SPINDLE_OCRA_TOP_VALUE (in hex) and SPINDLE_PWM_MAX_VALUE (in float) to be the same. 

I believe the CS bit flags are the ones responsible for setting the prescaler. The current one should be good for up to 2Mhz. Just change the top value to alter the frequency. 0x00FF and 255.0 should be right. 

Also as an FYI and this doesn't effect you, I don't remember why the top value is set at 1024.0, but I do remember there was a good reason for it. I'm pretty sure it had to do something with resolution.

timdavies91, 2018-08-27:
@chamnit make sense about being 1024 as guessing when I reduce it to 255 I'm going to loose some 0-5v resolution. The CO2 laser I'm using wants 20KHz but I'm hoping 8KHz will be okay, 1KHz seems to be no good, think it's overdriving the inverter as it shuts down mid run. If I need 20 KHz I'm thinking  I'll need to look at changing the prescaler due to resolution issues.

Thanks again will let you know how I go

chamnit, 2018-08-27:
@timdavies91 : You can try to disable the prescaler to have the PWM timer run at the full 16MHz, and set the PWM top value to 800. This should get you to 20kHz.