gnea/grbl-Mega #73: Z Axis Not Homing Correctly ?

NEMA34, 2018-09-02:
Hi , First off Thanks to GRBL Software Engineers for a great bit of Control Software 

My Set Up : I am Running GRBL 1.1f on a Mega 2560 - Was Programmed using the Hex File Method.

I Connect to GRBL through UGS ( Nightly Build Latest )

My Machine is a CNC Router : See the Photo Attachment of my Limit Switch Positions & my Cartision Positioning Setup.

HOMING ISSUE DESCRIPTION

I have tested the Limit Switches Manually to make sure they are all working in UGS in Verbose Mode & they appear to be working Just fine - Here is the Output 

[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0>
[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0>
[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0|Pn:Z> - Me Manualy Triggering Z Axis Limit Switch
[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0|Pn:Z|WCO:0.000,0.000,0.000>
[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0|Ov:100,100,100>
[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0>
[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0|Pn:X> - Me Manualy Triggering x Axis Limit Switch-
[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0|Pn:X>
[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0>
[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0>
[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0|Pn:Y> - Me Manualy Triggering Y Axis Limit Switch
[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0|Pn:Y>
[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0>
[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0|WCO:0.000,0.000,0.000>
[verbose]<Alarm|MPos:0.000,0.000,15.000|FS:0,0|Ov:100,100,100>

When i Run the Homing Cycle the Z Axis Moves slowly Upwards towards the Switch & I Hear the Switch Activate but the Z Axis Stepper Motor Continues to keep on Going up even though the switch has Triggered ?

So i Have to hit the Emegency Stop Panic Button to kill all power to the controller

My Limit Switches are wired ( Normally Open With 1K Pull Up Resistors to +5V & 100nf Ceramic Caps on the Grounds ) Making an RC Filter - I See no False Triggering in Vebose Mode.

My GRBL 1.1F Settings are as Follows 

Grbl 1.1f ['$' for help]
>>> $$
>>> $G
$0=10
$1=25
$2=0
$3=3
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=1
$21=0
$22=1
$23=7
$24=25.000
$25=50.000
$26=250
$27=20.000
$30=24000
$31=5000
$32=0
$100=159.426
$101=159.521
$102=100.000
$110=3000.000
$111=2400.000
$112=500.000
$120=25.000
$121=25.000
$122=10.000
$130=670.000
$131=1400.000
$132=200.000
ok
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
>>> $I
[VER:1.1f.20170802:]
[OPT:VNM,35,255]
ok

I Have Tried Re-Downloading the Hex File & Re-Flashing the 2560 Again - Still the same

Also i have put a mini scope on the Z Axis Limit Pin to see if there was any Noise Present, it is nice & clean.

Switch Untriggred it reads +4.870V (HI) & Triggred it Drops to 0.092V ( LOW ) with no apparent noise on the Z I/O Pin in ether Hi or Low ?

Is this a Bug with the Z-Axis or a config Error ?

Many Thanks in Advance

Dave528, 2018-12-15:
I was looking through your GRBL FW settings and setting $21 (Hard Limits) = 0 is incorrect for running homing cycles since this disables hard limits.  I would change this to $21 = 1 to enable the hard limits.  The $22 (Homing Cycle) setting is correct, but does not work with hard limits disabled.
NOTE: I've only used GRBL on the Mega328 platform, so if I'm mistaken about this please let me know.

beautifulsmall, 2020-08-20:
To confirm the workaround, 
Z switch showing in status but unresponsive during homing.
I had $3=5  for direction invert on the X and Z ,

I set $3=4 and re-wired the X stepper to drive the opposite direction and it Homed perfectly in all directions. 
had ,boolean :
$21  hard limits enabled =1
$20  soft limits disabled.  =0
 using NC switches with
 $5 invert on all limits 
 discussion suggests  X and Y direction inversion affects the homing switch polling.