gnea/grbl-Mega #74: This sketch compiles and uploads Grbl to your 328p-based Arduino!

sfinexer, 2018-09-20:
File
https://github.com/gnea/grbl-Mega/blob/edge/grbl/examples/grblUpload/grblUpload.ino
line 2:
This sketch compiles and uploads Grbl to your 328p-based Arduino!

harshjn, 2019-04-20:
Agreed! There is some problem in the comment probably. In any case, I have not been able to use this successfully on my Arduino Mega so far. Still Trying.

drf5n, 2022-03-13:
That's designed for UNOs with the 328p

For the Mega on my Mac,  I eventually found my way to this avrdude line:

    avrdude -c stk500 -C /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -P /dev/cu.usbmodem1433401 -p atmega2560 -D -U flash:w:grbl.hex:i

But you'd need to update the -C config file path and `-P /dev/xxxxx` for your device.