gnea/grbl-Mega #76: Losing Steps, Pulse Width>?

Ritzy22, 2018-10-06:
Hi,

Step pulses occasionally are longer, causing lost steps??

I have had an issue with inconsistent probe depths (probing the exact same spot multiple times) seeing variation from + or -0.004-0.028mm.

This issue is reduced significantly when 5 microstepping vs 10 microstepping however, it still occurs and i suspect it is due to there being half the number of steps required.

I have replace the UNO with a mega (both clones) as the UNO was using the CH340G chip and that seemed to help a bit however still plagues the 10 microstep making it unusable. I have also tried to reduce noise as much as possible screening all signal wires, strong pullups, screened USB cable and using my bench top power supply to no avail.

Looking at the step pulses it seems that occasionally they may vary in length getting longer (link)?

https://imgur.com/a/OjcnkzM

This occurs no matter what the pulse width is.

Is there something i'm missing???
Any help would be greatly appreciated :)

note: using the Geckodrive 213V

cprezzi, 2019-02-19:
@Ritzy22 You cannot expect to get more precision with microstepping! Only full stepps are defined positions, all microstepps are only somewhere between the full stepps, but not repeatable.

109JB, 2019-02-19:
In real life, microstepping does increase precision and microsteps ARE repeatable. A 1/16th microstep may not be exactly 1/16 of a full step, but definitely is repeatable.  Also, 1/2 steps are exactly 1/2 of a full step. 

While increasing the microstepping may not technically be "exact", a 1/16 microstep will be somewhere between 0/16 and 2/16, and 2/16 will be somewhere between 1/16 and 3/16 and so on. So while you cannot say that they are "exact" positions, they are definitely an improvement over no microstepping.

easytarget, 2019-02-19:
More importantly; the question being asked mentions micro-stepping, but I'm not sure how repeating old adages about how micro-steps are not real steps added to the discussion. OTOH; I'm guessing this is a lead-screw z drive, so maxing out the microstepping should not be a priority for the OP.

I am curious why you are lowering the pulse width so far @Ritzy22 ? the default is 10 microseconds and seems to work for most, Even at very high feedrates I've never found it a limit.