gnea/grbl-Mega #77: Adding LCD, Keypad, and SD card support

bdurbrow, 2018-10-06:
I'm thinking of setting this up on a common 3D printer platform (Arduino Mega 2560 + RAMPS 1.4 + LCD w/ SD card reader), and I'd like to make it run standalone (like Marlin does).

So, I figure I'm going to need to write/port a SD card reader module, keypad module, and LCD menu system module. Not a big deal, I'm a programmer. :-)

However, I would like some input from the official developers as to where I should hook into grbl... without making a mess of things.

So... where should I inject the G-code I have read off of the SD card? Where should I drop a call to my LCD update routine; and where should I drop a call to my keypad polling routine?

I'm going to want to support a Manual Pulse Generator hand wheel also... where is the proper place to inject jogging data? Or should I take over the Arduino when in MPG jog mode, and then update the machine position global variables afterwords (and if so, which variables should I update?)

Also, is there an officially sanctioned way of directly setting (from a C function, not by injecting a G-code into the input datastream) feedrate override, spindle override, and coolant override parameters?

Any pointers you can provide (er, no pun intended) would be much appreciated. Thanks!

terjeio, 2018-10-15:
I am using function pointers to achieve [a similar goal](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack). But perhaps not the kind of pointers you are after...? Another kind of pointer may be this [ESP32 port](https://github.com/bdring/Grbl_Esp32) as it has SD card support, and perhaps it is possible to use its serial comms over blueetooth for GCode injection?

dangerousderek, 2018-10-15:
I’ve been scouring the web looking for exactly this. I’m no coder, but I love grbl and I just want a stand alone system using a ramps board for a laser cutter I built. Any progress you make on this front would be awesome and I would be happy to beta test. Thanks!!!

bdurbrow, 2018-10-16:
> scouring the web looking for exactly this

So was I. I started trying to modify MK4Duo (a Marlin variant) to remove the 3D Printer specific components, but have determined that they are too deeply embedded into the firmware for that to be practical.

So I'm doing this instead. So far, I've got the device drivers (SD, Matrix Keypad, HD44780 LCD) ported from the Arduino codebase; and partially integrated into the Grbl codebase (there's quite a bit of incompatibility between the base Arduino core and Grbl, in particular in use of system resources like timers; so I've had to re-code quite a bit of the low-level Arduino support libraries to make it compatible... it wasn't a matter of drag-and-drop-and-compile).

Unless/until I'm told otherwise, I'm injecting the data from the SD card and jogging into protocol_main_loop(); and have put a call to my UITask() function into protocol_execute_realtime() and protocol_exec_rt_suspend().

Developkings, 2018-10-18:
i believe using a ramps friendly premade solution like the lcd 2004 or ultra would be the best solution. i dont know about coding that much, but would it be easier to adapt marlins integration of this components t grbl instead ? if i`m not wrong it already as a basic cnc menu on it

bdurbrow, 2018-10-18:
> but would it be easier to adapt marlins integration of this components t grbl instead

No, it's not. I already went down that road.

> ramps friendly premade solution like the lcd 2004 or ultra

Hardware wise, that's what I'm doing. I'm using the Reprap Discount Smart Controller with a RAMPS 1.4 board. I'm not making my own hardware for this project - the whole idea was to use off-the-shelf parts.

For performance reasons, I'm using the character LCD, and not the graphic LCD - I'm hoping to get nice, quick DRO updates during jogging.

bdurbrow, 2018-10-27:
Status Update:
Drivers are working, but could stand some optimization (reviewing the drivers that Arduino ships, they are excessively abstracted in their construction; and use slow IO functions. I intend to fix this before posting any code).

SD support is integrated into protocol.c; but is as of tonight untested.
Keypad support is fully functional.
Rotary encoder (i.e, the knob on the LCD pcb) support is fully functional.
LCD support is functional.

User Interface code is about half done. The UI Page system infrastructure is functional; the menu system is itself functional but needs more menu items; and the SD directory access is working (however, at the moment, nothing happens when you select a file - that code needs to be written yet).

Still to do:
SD file opening when you select a file.
Jogging.
Probing.
Part Zero setting.
MDI.
"Conversational" operations for common simple tasks that don't warrant going to the extent of firing up a CAD/CAM program (drill a hole, bolt hole pattern, cut a slot, machine a face flat, etc).

Jogging, probing, and part zero setting basically just involves sending the appropriate command to the main grbl routine; in the same way that a PC would initiate that operation. The infrastructure is in place for this, it just needs to be "wired up" to the keypad & menu system.

MDI requires implementing a text editor to enter the MDI command line on.
Conversational requires a similar text editor, but with a "form" style layout.

dangerousderek, 2018-10-27:
I can't believe how much progress your making! It seems like such a no brainer I am amazed that no one has done it already, in fact, I kind of figured it must be extremely difficult to do since it hadn't been done. You're the man. Keep us updated and as soon as your looking for some guinea pigs I call dibs! I am pretty active in a couple of other forums and groups having to do with DIY CNC stuff and those folks are going to freak out if you manage to pull this off!

bdurbrow, 2018-10-28:
I have now achieved my primary objective: GRBL, on a mega2560 with a RAMPS 1.4 & RepRap Discount Smart Controller, is streaming g-code from a file stored on a SD card, selected via LCD, and performing updates to the display faster than the liquid-crystal media can physically respond. Oh, and it's doing it while maintaining responsiveness to serial commands, too...

There's still additional work to be done, but as of right now, I'm pretty pleased with the way it's working. 😎

dangerousderek, 2018-10-29:
Are you going to fork it from this github? I need it so bad! You’re a super hero. Good job!

bdurbrow, 2018-10-29:
> Are you going to fork it from this github?

Yes.

MDI is working, and the UI for jogging is halfway there. I need to go dig out a MPG handwheel and cook up a cable for it, but that's probably going to have to wait for tomorrow.

Anyone who wants to test this should probably make sure they have the requisite hardware now, since it often takes weeks to months for shipping...

I suggest:

- A suitable SD card. (Anything that works with other Arduino SD software/hardware - like a 3D printer - should work... I'm using SD drivers originally sourced from the Arduino codebase). Make sure you format it correctly, though, with the tool from the SD Association (https://www.sdcard.org/downloads/formatter_4/index.html). I've had mixed luck using the built-in formatting tools provided by the OS (both OS X and Windows) on SD cards... sometimes it works, sometimes it doesn't. Using the tool from sdcard.org seems to avoid this issue.
- A mega2560.
- A RAMPS 1.4 w/ stepper drivers.
- A RepRap Discount Smart Controller LCD interface w/ Smart Adapter (the one with the 20x4 character LCD, rotary encoder w/push button, extra push button and SD card slot; NOT the ones with a graphic LCD, this firmware doesn't - and won't - support those).
- A 4x4 matrix keypad (there are inexpensive membrane-style ones available on eBay, search for "Arduino keypad 4x4" and pick one - or 10 - that's cheap; and if you get the same one that I got, I've got a PDF file that can be printed out and glued over the keypad to show what the key functions are)
- A suitable MPG handwheel. (The ones I'm using are 100ppr, 60mm handwheels, and can be found on eBay by searching for "CNC MPG 5v 60mm"; at the moment they are under $20usd).
- Switches for Cycle Start and Feed Hold (I'm using some arcade-style push-buttons, again sourced from eBay) and 5K pots for feed rate override and spindle override. These components will require soldering, so make sure you've got a suitable soldering station (temperature controlled iron, "helping hands" to hold stuff, magnification, good solder and flux, etc).
- Suitable cabling to attach everything. (I ordered some long male-to-female 0.100in "DuPont" style hookup wires, and some 1x2, 1x3, 1x8, 2x4, and 2x5 "DuPont" style shrouds from eBay; also I had on hand some suitable ring terminals for hooking up the MPG handwheel).
- And, of course, your CNC machine itself. 😁

dangerousderek, 2018-10-30:
I’ve got everything but the 4x4 pad and the mpg wheel! I’m stoked!

bdurbrow, 2018-10-30:
Well... that's a bummer. I can't find the MPG handwheels I had here... I already had a few spares on order, but they're gonna take a while to get here. I guess I should order a few more spares now, also... (of course, you know that just as soon as the new ones show up, the old ones will magically appear, also...)

So I'm having to use the (much coarser resolution) encoder on the LCD for now...

🤨

dangerousderek, 2018-10-31:
Are you building an enclosure to hold all this biz? If so can I get some pics? I have a feeling you’re doing exactly the sort of stuff I need to do with my rig...

dangerousderek, 2018-11-11:
So.. I hate to nag, but any updates? Thanks a ton!

bdurbrow, 2018-11-13:
Sorry about the delay... I've had to wait on parts arrival - the MPG handwheels I had are still AWOL; and I had to order replacements. However, the new ones have arrived and I've made quite a bit of progress:

MPG Jogging works now, and I've not been able to over-run the CPU with pulses (I'm using an interrupt to capture the input pulses; the polling loop can, at high speeds, get swamped).

The infrastructure is in place for the conversational operations; and drilling is in place. Other operations (facing, hole interpolation, etc) need implementing.

MDI works.
Part Zero setting works... but there's a feature I want to add (to take into account tool diameter and touch-plate thickness).
Probing should be working, but requires testing.

I'm hoping to have a pre-beta up on github soon... perhaps a week or two.

dangerousderek, 2018-11-13:
You don't have to apologize! I am just super excited and I am having a hard time reigning it in. I look forward to testing the pre-beta! It sounds like you're going above and beyond the call of duty!

bdurbrow, 2018-11-19:
Getting real close...

manisaranbala, 2018-11-19:
https://sites.google.com/view/arrsoft/автономные-контролеры/автономнй-контроллер-для-
фрезера?authuser=0

bdurbrow, 2018-11-19:
@manisaranbala 

Um... I don't read Russian, and google translate is making hash out of that.

Is that supposed to be some sort grbl based controller?

manisaranbala, 2018-11-20:
http://www.cnc-club.ru/forum/viewtopic.php?f=41&t=20190

manisaranbala, 2018-11-20:
https://sites.google.com/view/arrsoft

bdurbrow, 2018-11-21:
OK, here's something to play with.

WARNING: This is pre-beta software. This has not been tested with motors attached yet (my machine is in pieces at the moment). Although it seems to be working on my bench when connected to my oscilloscope, it could crash your machine! So, be careful... and be ready to pull the plug or hit your emergency stop (whichever is more applicable to your situation).

Also, there are known issues with the conversational programming... so don't try that yet.

I have included some photos of the wiring required for the keypad and MPG handwheel in the "UI Support" folder. A PDF file of the keypad overlay is in there also (it has versions in it to support 4 and 5 axis machines, even though grbl itself only goes to 3 at the moment).

Note that unlike a PC, you don't hold down the 'Shift' key; it's a toggle. Push shift, _then_ push the other key.

Probing and setting part-zero are done from the Jog page.

Be aware that editing the tool table does put a cycle on the EEPROM of the atmega... so you probably want to set up your tools once and leave it alone. So does setting part-zero from the Jog page (I intend to implement a settings menu later on that will allow this to be disabled, so that you're not burning EEPROM cycles every time you run a part).

Documentation still needs to be written.

And yes, eventually this will get a proper github repository. I just wanted to get something up here for you all to play with.

😀

[grbl-Mega.zip](https://github.com/gnea/grbl-Mega/files/2605324/grbl-Mega.zip)

dangerousderek, 2018-11-24:
I’m super excited to mess with this. With the holidays I’ve been to busy to experiment in the shop, but I’m hoping this week to give it a try. One quick question- I still don’t have an mpg wheel (on a boat from China) will the rotary encoder on the reprap discount full graphic lcd work? I’m super stoked to try all the awesome built in features you’re working on, but for now I would settle for being able to see the status of the machine and do basic moves on the axis’s. Thanks again, you’re the man.

bdurbrow, 2018-11-25:
Yes, the MPG is optional equipment. Recommended, but optional.
Just disable (by commenting it out with a // )this line in config.h:
#define USE_UI_ENCODER_B

This will redirect the jogging inputs from the MPG inputs to the main UI knob.

Minimum hardware requirements for using the UI:
Arduino Mega 2560 (obviously... :-)
RAMPS 1.4
Reprap Discount Smart Controller 20x4 LCD & rotary encoder panel.
4x4 Matrix Keypad

RunnyBore604, 2018-11-26:
Here's a solution for laser burners/engravers:
https://openbuilds.com/threads/arduino-based-off-line-cnc-controller-for-laser.13315/

bdurbrow, 2018-11-26:
@dangerousderek - 

> on the reprap discount full graphic lcd work

Uh oh, I hope that's a typo... this does **_NOT_**, and will not, support **_GRAPHIC_** LCDs.
20x4 HD44780-compatible Character LCDs **_ONLY_**.

If you need to order one, and don't want to wait for the slow boat from China; Amazon has them with Prime Shipping for about $12USD... on eBay they can be had from China for under $8USD at the moment.

For example:
https://www.amazon.com/BIQU-Display-Controller-2004LCD-Adapter/dp/B01FLU9X5Q/

@RunnyBore604 -
It looks like that's not open source?

dangerousderek, 2018-11-26:
@bdurbrow I think I’ve got one of those laying around. I was initially planning to use one of these:
https://www.amazon.com/KINGPRINT-Graphic-Display-Controller-Adapter/dp/B076WQQX5K/ref=mp_s_a_1_1?ie=UTF8&qid=1543270471&sr=8-1&pi=AC_SX236_SY340_QL65&keywords=reprap+full+graphics+display

But I failed to read the 20x4 detail in your BOM! Thanks!

easytarget, 2018-11-29:
Sorry OT ; but not sure how best to address this..

> @RunnyBore604 -
> It looks like that's not open source?

He's cross posting/promoting from his own OpenBuilds project here, (check user details). Over there he says he wont share the source code.. Kinda against the ethos, and if he's also lurking here then I imagine GRBL is involved, GPL etc.. Anyone got a Openbuilds account and want to flag him?

Bazza1959, 2018-12-22:
Love your work bdurbrow, as soon as I came across this page I immediately dug out the ramps and mega stuff i purchased over a year ago and got all excited and started uploading your work, but uh oh, when I reread everything I found that I have one of those graphics lcds, so now have to wait for slowboat, as I am in ozzieland, but not to worry too much as I can still be printing all the components for the cnc in the meantime, thank you very much for all your hard work, I really do appreciate it, Barry

BillEng, 2019-01-10:
Thank you bdurbrow, just what I have been wanting.
I have tried this on hardware that has previously had Marlin on  it (MPCNC version), I have no stepper drivers, no keypad and no second encoder.
Connecting with bCNC I can run a gcode program, love the real time DRO, the LCD updates with feed and spindle overrides and preset Gcodes adjusted in bCNC but the push button does nothing, I can turn to different menu pages also SD card not recognized, tried two one from my 3D printer (8gig) and one I use on the MPCNC (16gig).
I have tried commenting out the second encoder as mentioned above, no change. 
Am I missing something just running without anything else attached to Ramps.
 Thanks again, I can wait.

BillEng, 2019-01-14:
Well I feel a right plonka with my comments above I had read the post a 100 times but obviously not absorbed them, I think I'll go dig a hole and bury myself.
I have added a keypad and it is working to the point of development.
The encoder on the LCD is too sensitive, can this be tweaked.
 I could buy a second encoder that has twice the PPM of the one mentioned above at 400 PPM could this be accommodated.

Bazza1959, 2019-01-17:
your not the only one, I am a plonker also, after purchasing the required components I have just realised you guys are setting up a lathe or a milling machine where as I am doing a cnc router is there anyway I can still setup the keypad and lcd screen for my setup and if so what do I need to do before I go dig that hole

BillEng, 2019-01-20:
It's grand to know I'm not alone in the world Bazza1959.
I'm using a CNC router, as yet I can see that it would suit my needs, just ignore the features that are not needed or maybe (a big maybe in my case) mod the menu.
As mentioned above at the moment I am using Marlin, no disrespect to the devs but I hate all the 3D printer stuff in the menu.
I have a remote pendant that helps but can only jog in increments, my wish list would be.
Manually using the machine, handwheel.
Feed rate override while running a gcode.
Run gcode from SD card, semi headless.
To enter gcodes and/or some preset gcodes from a keypad.
The developmet so far seams to fit the bill.

bdurbrow, 2019-02-09:
Sorry about the delay, I just haven't had much time to work on it lately. I should have something new up soon though...

@BillEng - The encoder on the LCD can be adjusted somewhat by changing it's configuration (UI_ENCODER_A_MODE in cpu_map.h); but it was designed for the encoder on a RepRap Discount Smart Controller; if you are using something else with more pulses per detent then I'll have to add a sub-divider (not a big deal; I just need to know to do it).

@Bazza1959 - I'm intending this to be usable on both mills and routers. If you can live without threading, a lathe could be setup also... but at the moment grbl has no support for syncing the axis movement to the spindle.

BillEng, 2019-02-10:
bdurbrow, please don't retire you will have less time (-:, don't do any mods for an encoder at this time, I asked because I could buy one from my home country (UK) without the hassle of direct from China.
When I was trying the firmeware I did have an odd problem, the second row of buttons is a duplicate of the first row, checked through the files as best I could.
Flashed a test prog onto the Mega to test the keypad and my extension lead, all OK, I even tested the continuity of the Ramps tracks from AUX2 to where they connect to the Mega again all OK.
Thanks again.

cadenote, 2019-02-26:
Great work @bdurbrow ! Thank you very much. It is what I was searching for.
When I have played with it, I have noticed that files coming from gcodetools ( inkscape )  stop when encounter a line with only a comment or a void line in the nc file.
To overcome this issue I have introduced minor changes in protocol.c and SdFile_grbl.cpp files.
I hope this will be useful and not to spoil anything.
[grbl_Mega_proposal.zip](https://github.com/gnea/grbl-Mega/files/2905655/grbl_Mega_proposal.zip)
x:)

Bazza1959, 2019-02-27:
@cadenote is there anything needed to be done with the files if used on a cnc router or can I just simply replace the old files with the new ones T.I.A.
Barry

cadenote, 2019-02-28:
Just replace them in libraries/grbl directory and upload again grblUpload.ino into your arduino.
:)

Bazza1959, 2019-02-28:
BillEng if using an arduino mega and a ramps 1.4 board do not connect power to the ramps board without having stepper motors connected, Th reason is that it will burn out the stepper drivers and maybe even the ramps board, ask me how I know this lol very frustrating, it took me a month to work out why my steppers wouldn't move and another month to get new drivers from china and now that they have arrived I can see if the boards are stuffed also, but I did order new ramps board and an uno cnc board as well and they should be here in a few days

sinfocomp, 2019-03-15:
Hello, I'm trying but can't load any file, what is the correct file extension to use?

cadenote, 2019-03-16:
> Hello, I'm trying but can't load any file, what is the correct file extension to use?

I also had the same problem. I had to dig into the software to find that extension has to be .nc or .NC

sinfocomp, 2019-03-20:
can you please let me know how to load a file, i used .nc and .NC but still my display shows NO FILE

Bazza1959, 2019-03-21:
sinfocomp what file are you trying to load we are not sure of what you mean exactly. Are you trying to load cadenotes file or the original GRBL file from bdurdro or what?

sinfocomp, 2019-03-21:
I have loaded firmware however
I have two problems
I cannot compile without the mpg
And I get NO FILE on lcd even if my SD card contains a testjob.nc file
Thanks

Bazza1959, 2019-03-21:
you should be able to find and comment out the mpg wheel so that fixes your first problem but you will have to wait until bdurbro can answer the second one. I am not at the stage of needing this firmware as I am still waiting on a new ramps board and drivers as I stuffed my old ones

sinfocomp, 2019-03-21:
I commented the mpg line as noted somewhere on this thread however I get compile errors unless I Un comment the mpg line

cadenote, 2019-03-21:
@sinfocomp  in order to see your file in the display you have to presss menu key and after that you have to select SD Card option. Regards.

sinfocomp, 2019-03-21:
Cadenote thanks, and in order to not use an mpg? Just one line needs to be commented on source?

cadenote, 2019-03-21:
Well, up to I guess, to compile without errors you need to comment  out two lines in config.h : 
#define USE_UI_ENCODER_B
#define  USE_UI_ENCODER_B_ISR
However in that case jog mode has to be tested deeply.

sinfocomp, 2019-03-26:
Hello, i need some help making this work on a MKS Base V1.5 board,
my problem so far is the keypad pins,can someone please helpme on remaping pins to A11,A12,D11,D12, and A13,A14,A15,D0,  Thanks in advance

sinfocomp, 2019-03-29:
Got a ramps 1.4 board for testing and up and running, any information on where the menu text is, i would like to translate to spanish.
Thanks

cadenote, 2019-03-30:
Mainly they are in UIMenus.cpp but  also there are lots of messages in report.c
Regards,

sinfocomp, 2019-03-31:
> Hello, i need some help making this work on a MKS Base V1.5 board,
> my problem so far is the keypad pins,can someone please helpme on remaping pins to A11,A12,D11,D12, and A13,A14,A15,D0, Thanks in advance

any help on re asigning the keypad pins ?

Bazza1959, 2019-04-01:
sinfocomp I feel you are asking the wrong people by being on this site, the title of this is GRBL-Mega.
the MEGA stands for an Arduino Mega 2560 board with a Ramps 1.4 cnc shield on top, and I am unsure that any of our programming would work on boards that are not the same. Maybe some of the other guys can correct me if I am wrong as I am not an expert at it, which is why I am on this site and I sincerely apologise if I am wrong

Bazza1959, 2019-04-01:
OUCH $55 aussie just for the board, my mega and cnc shield was less than $4 delivered plus another few for the drivers. Sorry but I have no idea on how to remap pins
Footnote: still waiting for new ramps boards after blowing mine up

sinfocomp, 2019-04-10:
finaly got it working !!!!
just dont seem to know where to hookup feed hold button ???
looked at cpu_map.h but feedhold is analog 9   and looks like is being used for keypad, anyone here knows where to conect feed hold ?

sinfocomp, 2019-04-20:
please help, where do i connect the pot for feed overide ?

lgojniczek, 2019-05-03:
Hi makers

I would like to ask how to clone the Y axis? Is it possible ?

I have cnc router that have two stepper and 2 motors on Y axis - one is reversed in rotating

Thank you for help

zvizdic, 2019-05-03:
(Minimum hardware requirements for using the UI:
Arduino Mega 2560 (obviously... :-)
RAMPS 1.4
Reprap Discount Smart Controller 20x4 LCD & rotary encoder panel.
4x4 Matrix Keypad)
Can I use external drives ?

JoeAndrew, 2019-05-12:
> > Are you going to fork it from this github?
> 
> Yes.
> 
> MDI is working, and the UI for jogging is halfway there. I need to go dig out a MPG handwheel and cook up a cable for it, but that's probably going to have to wait for tomorrow.
> 
> Anyone who wants to test this should probably make sure they have the requisite hardware now, since it often takes weeks to months for shipping...
> 
> I suggest:
> 
> * A suitable SD card. (Anything that works with other Arduino SD software/hardware - like a 3D printer - should work... I'm using SD drivers originally sourced from the Arduino codebase). Make sure you format it correctly, though, with the tool from the SD Association (https://www.sdcard.org/downloads/formatter_4/index.html). I've had mixed luck using the built-in formatting tools provided by the OS (both OS X and Windows) on SD cards... sometimes it works, sometimes it doesn't. Using the tool from sdcard.org seems to avoid this issue.
> * A mega2560.
> * A RAMPS 1.4 w/ stepper drivers.
> * A RepRap Discount Smart Controller LCD interface w/ Smart Adapter (the one with the 20x4 character LCD, rotary encoder w/push button, extra push button and SD card slot; NOT the ones with a graphic LCD, this firmware doesn't - and won't - support those).
> * A 4x4 matrix keypad (there are inexpensive membrane-style ones available on eBay, search for "Arduino keypad 4x4" and pick one - or 10 - that's cheap; and if you get the same one that I got, I've got a PDF file that can be printed out and glued over the keypad to show what the key functions are)
> * A suitable MPG handwheel. (The ones I'm using are 100ppr, 60mm handwheels, and can be found on eBay by searching for "CNC MPG 5v 60mm"; at the moment they are under $20usd).
> * Switches for Cycle Start and Feed Hold (I'm using some arcade-style push-buttons, again sourced from eBay) and 5K pots for feed rate override and spindle override. These components will require soldering, so make sure you've got a suitable soldering station (temperature controlled iron, "helping hands" to hold stuff, magnification, good solder and flux, etc).
> * Suitable cabling to attach everything. (I ordered some long male-to-female 0.100in "DuPont" style hookup wires, and some 1x2, 1x3, 1x8, 2x4, and 2x5 "DuPont" style shrouds from eBay; also I had on hand some suitable ring terminals for hooking up the MPG handwheel).
> * And, of course, your CNC machine itself. grin

I've tried the code and it works, without steppers.
Is there a wiring diagram of where to wire exactly? I think about the buttons the most.
In the meantime, I try to merge your projects with the lathe sync code.

[73 more replies]