gnea/grbl-Mega #82: Installation in Arduino 1.8.7 UI

macdroid53, 2018-11-13:
I can't seem to get the mega version to override the grbl files in that come with the Arduino 1.8.7. This is on Ubuntu 18.04. If I attempt to build I get pages of warnings like this:

`In file included from /home/mac/Arduino/libraries/grbl/grbl.h:43:0,
                 from /home/mac/Arduino/libraries/grbl/examples/grblUpload/grblUpload.ino:27:
/home/mac/Arduino/libraries/grbl/nuts_bolts.h:56:0: warning: "max" redefined
 #define max(a,b) (((a) > (b)) ? (a) : (b))
 ^
In file included from sketch/grblUpload.ino.cpp:1:0:
/home/mac/Downloads/Arduino/arduino-1.8.7/hardware/arduino/avr/cores/arduino/Arduino.h:93:0: note: this is the location of the previous definition
 #define max(a,b) ((a)>(b)?(a):(b))
 ^`

chamnit, 2018-11-13:
You need to follow the compiling instructions exactly. Grbl hacks how the Arduino IDE works to allow compiling of Grbls custom firmware.

closed 2018-11-13 by chamnit

macdroid53, 2018-11-13:
Sorry...please point me to "the compiling instructions". Because I got it working on one pc, but not this one and I can't find "compile grbl this way for mega 2560" instructions.

macdroid53, 2018-11-13:
Ok, from that, (now bookmarked, thank you!), link: 

"NOTE: If your environment is clean and Arduino IDE compiler still throws "warning: [...] redefined" messages, you may need to uncheck File -> Preferences -> "Aggressively cache compiled core"."

Once I unchecked that it compiles fine. Apparently, something was cached. :(

macdroid53, 2018-11-13:
PS, is there a grbl specific forum somewhere? I'm having some other issues with limit/reset inputs I'd like to discuss...and github probably isn't the right place.