gnea/grbl-Mega #85: Continous toggle into Hold state

macdroid53, 2018-11-18:
I now have 1.1f installed on my Mega.

With the UGCS gui open grbl continuously toggles into Hold state.

If I hit Cancel or Resume it toggles to Idle, then immediately back to Hold.

If I just watch (i.e. click nothing) it will toggle to Idle then back to Hold. This repeats continuously.

macdroid53, 2018-11-18:
Just replaced the mega with an UNO, uploaded uno version of 1.1f and it works fine...

chamnit, 2018-11-19:
There isn’t anything that would cause this issue in Grbls side, unless there was some kind of EMI or waiting fault in the hold and resume input pins. Check those pins for any issues.

macdroid53, 2018-11-19:
I checked all limit inputs and the reset input (nothing connected to the rest) with my o'scope and found no glitches or appreciable noise spikes.

It's a well used Mega so maybe it's just tired. As I said, I replaced it with an Uno and all seems to work fine. Gcode files that just went wonky on the Mega, now work fine.

chamnit, 2018-11-19:
K. Thanks for the feedback. I'll keep an eye out on any other random instability reports like this one, but you're probably right that it's your Mega being wonky.

closed 2018-11-19 by chamnit