gnea/grbl-Mega #87: MkS 1.4 + tmc2130

CorroSamuele, 2018-11-24:
I'm creating a cnc with mks 1.4 and tmc2130 I loaded grbl for mega but the engines do not move, I use cncjs, the program reads the card and its movements but the engines do not

dudman, 2018-12-29:
I've got the same, did you get any info, or anything to work ?

CorroSamuele, 2018-12-29:
You must remove all jumper under driverInviato dal mio dispositivo Huawei-------- Messaggio originale --------Oggetto: Re: [gnea/grbl-Mega] MkS 1.4 + tmc2130 (#87)Da: dudman A: gnea/grbl-Mega CC: CorroSamuele ,Author I've got the same, did you get any info, or anything to work ?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/gnea/grbl-Mega/issues/87#issuecomment-450499457",
"url": "https://github.com/gnea/grbl-Mega/issues/87#issuecomment-450499457",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]