gnea/grbl-Mega #88: Soft wall

hpirila, 2018-12-12:
I'm thinking of new functionality to grbl which would be helpful in some operations made in jog mode. 

I call this new functionality a soft wall. It would be similar to already implemented functionality soft limit. In soft wall user can set and unset limits in the axis using G commands. When machine approaches or reaches soft limit it slows down and stops similar way as with soft limit, but does not set any alarm or hold.

I'm interested to hear comments and ideas how to implement this or is this already available. Also real help in programming would be appreciated as I haven't done any grbl mods before.

109JB, 2018-12-12:
IMHO, not needed in Grbl, but could be added to an interface program. Would be better suited there.

chamnit, 2018-12-12:
I thought I setup Grbl’s jogging mode to do this already. Can’t remember for sure but I think jog motion outside the machine space just error out or get ignored. I would need to check if it’s there already and how I did or didn’t do it.

hpirila, 2018-12-12:
Thanks. I would prefer to do it in interface program as well. 

But does current grbl soft limit implementation make me a box window which I can't jog out, example I am in position X0 and I would like to make soft limit to both X10 and X-10.

hpirila, 2018-12-12:
Further trying I can see error15 "Jog target exceeds machine travel. Command ignored." when I try to jog beyond soft limit. 

This is difficult to handle with interface program. It would be easier if grbl allows jog commands beyond soft wall but then just don't allow movement beyond.

chamnit, 2018-12-12:
I disagree. It’s better to have Grbl only move to targets when there is a valid move. You can get yourself into much more trouble otherwise. It’s not hard for an interface to deal with soft limits.

closed 2018-12-12 by chamnit

109JB, 2018-12-14:
Indeed. The interface I wrote does exactly that. It calculates the distance from the current position to the appropriate travel extent and issues a jog command to that distance and not further.

hpirila, 2018-12-14:
I will implement something similar to my interface program. It will stay between the 'walls' in 95% of the cases. Problem comes example when jog cancel is requested and jog direction changed immediately after. Interface can't know the current location exactly, while grbl can of course. That is not really major problem as I can inform user soft wall is not active until you stop for a moment to sync again from the status report.