gnea/grbl-Mega #92: Using grbl with other MCUs

PerThomas, 2019-02-27:
Hi
I wonder if anybody have an idea of whether or not  it is a lot work to port grbl to be used with other MSCs. I considering  using grbl on a project that uses an STM32 micro-controller to control some stepper motors.

Im an experienced programmer myself, but would appreciate a few pointer on where to start

PerThomas, 2019-02-27:
I meant MCUs

langwadt, 2019-02-27:
it's quite straight forward I've ported it to STM32. basically; reimplement serial.c, eeprom.c, the atomic access, fix all the GPIO access and setup a timer for the step interrupt 

https://github.com/langwadt/grbl_stm32

chamnit, 2019-02-27:
It’s not too bad. But I am working on an abstraction to make it easier. Been busy but intend to release it eventually.

PerThomas, 2019-02-28:
Thanks a lot for quick reply, I will have a look 😊

best regards

Per Thomas Hille

When you communicate with us or otherwise interact with Semcon, we will process personal data that you provide to us or we collect about you, please read more in our Privacy Policy<https://semcon.com/data-privacy-policy/>.

PerThomas, 2019-02-28:
Thanks a lot, I will have a look 😊, Im sure I will figure out something with the hints  below.

I will let you know how it goes.

best regards

Per Thomas Hille

https://github.com/langwadt/grbl_stm32

[https://avatars3.githubusercontent.com/u/1613189?s=400&v=4]<https://github.com/langwadt/grbl_stm32>

GitHub - langwadt/grbl_stm32: grbl for stm32 with nucleo stepper drivers<https://github.com/langwadt/grbl_stm32>
github.com
grbl for stm32 with nucleo stepper drivers. Contribute to langwadt/grbl_stm32 development by creating an account on GitHub.

bdring, 2019-02-28:
I have ported it to 2 other MCUs. It was pretty easy because the code is so well written and easy to follow.

- [PSoC5](https://github.com/bdring/PSoC_Grbl) (arm) This was was very easy and I did it over a long weekend.
- [ESP32](https://github.com/bdring/Grbl_Esp32) This was a bigger project and probably took a couple months of part time work.

My advice is to work backwards from the peripherals.  Get all the timers and PWMs working and the rest is pretty easy.