gnea/grbl-Mega #94: It is possible to use an analog input to control the forward speed, like?

isinet, 2019-03-15:


codemakeshare, 2019-05-31:
Hi, 
many things are possible :) I actually needed this feature myself for a project, and started experimenting a bit. I've got something working, tested on a RAMPS board, with a potentiometer connected to ADC3 (middle pin on ADC, other two pins on GND and 5V). The way I did it will ignore the feedrate override commands, as it continuously sets the override to the analog value. I also added a feature that a HOLD command is issued when the knob is at zero, and resume when it is increased again (otherwise it will keep creeping forward at the minimum feedrate).

You can find my experimental code here: 
https://github.com/codemakeshare/grbl-Mega/tree/edge_analog_feedrate
(branch edge_analog_feedrate - based off today's edge branch from gnea). 

This is the specific commit if you want to look at the changes/additions:
https://github.com/codemakeshare/grbl-Mega/commit/ae809b83a6e0efa27c3572b34e4cf12c4ffcec0c
You can customise the analog input to use in config.h: #define FEED_OVERRIDE_ANALOG_IN 3
The crucial changes are in protocol.c, line 247-262.
I hope it works for you. I've only done very minimal testing so far, so be cautious.

isinet, 2019-06-02:
Thx, I probe the next week and report thx