gnea/grbl-Mega #97: grbl ramps 1.4 limit switch

lols, 2019-04-28:
Hi, is the limit switch broken on ramps1.4?
I can home correctly, but when i accidental crash to the limit switch, it did not stop but keep going
How can i fix it?

swarfer, 2019-04-29:
limit switches have not been implemented for RAMPS.
Read all about it in config.h

moonface1977, 2019-05-18:
I have read the file and I can't find information about this. If The limit switches aren't implemented yet, which board should I use with the mega/ Thanks a lot for your answer.

coilheaddesigns, 2019-07-01:
Setup your soft limits then grbl will know where it is once you home machine.

fra589, 2019-07-02:
Hi all,  
Hard limits will be implemented soon on [grbl_Mega_5X](https://github.com/fra589/grbl-Mega-5X), there is an already working prototype.
I just need a little bit more time to finish the tests and to publish the version...  
Stay connected :-)  
@++;  
Gauthier.