gnea/grbl-Mega #100: Maximum allowable value for $100,$101 and$102 parameter (step/mm)

tceccarini, 2019-05-19:
I would like to ask what is the maximum allowable value for $100,$101 and$102 parameters (step/mm).
if I set all 3 to 12800 the controller crash after issuing $X, G91, G0 X10Y10Z10  commands, if set all to 8192 it restart.. more or less the same with 6400.

I hope 3200 will be a safe value...

Thanks

langwadt, 2019-05-19:
you have to take $110,$111,$112 into account, code mentions a maximum ~30kHz step rate

tceccarini, 2019-05-19:
Thank very much for your fast reply.

So the constraint is MaxRate[mm/min] * StepForMM[step/mm] <= 30000*60

So setting parameters with: 
$100 = $101 = $102 = 12800
$110= $111 = $112 = 130

satisfies the constraint, in fact, the controller doesn't crash any longer.