gnea/grbl-Mega #106: CLONE AXIS

coilheaddesigns, 2019-07-01:


coilheaddesigns, 2019-07-01:
Hey guys using ramps 1.4 need to clon Y axis to extruder, is this possible if so how ?

all advice appreciated.

fra589, 2019-07-01:
Hi @coilheaddesigns,  
Take a look to my [grbl-Mega-5X](https://github.com/fra589/grbl-Mega-5X) fork wich is able to do this.  
@++;  
Gauthier.

coilheaddesigns, 2019-07-01:
Hey man can you check issue #77 there is a version there with a working lcd.
Is there a possibility that you can throw that compatibility in to your version?

I really like the fact that the lcd gives updates on gcode. Functionality of the controller not that important to me. 
And your version with multiple additional axes is awesome.

I do believe that if you were to merge your configs we would be having a grbl for ramps that will be legendary.

closed 2019-07-02 by coilheaddesigns