gnea/grbl-Mega #107: not working with mega2560 + cnc shield v3

darkworks, 2019-07-19:
these firmwares are pretty annoying tried multiple firmware but none of them actually works  , i have connected mega 2560 + cnc shield + 3 stepper motors nema 17 and uploaded this grbl mega but its not moving the motors when i send gcode , 
can u tell me where is the configuration file of this grbl i mean there should be some sort of config file where u mention  step , dir and enable pins of your board into it so that it work ,

thanks

chamnit, 2019-07-19:
Grbl on Mega has a different pin configuration that isn’t compatible with Arduino Uno shields. See the wiki for the pin outs and you’ll have to find a way to connect them correctly.

closed 2019-07-19 by chamnit

drf5n, 2022-03-13:
Oh my-- it is very different.  The CNC Shield V3 schematic from https://blog.protoneer.co.nz/arduino-cnc-shield/ has the step pins to the Mega's 2,3,4, which are PE4,PE5,PG5, which aren't on the same port, etc... The definitions in   https://github.com/gnea/grbl-Mega/blob/df87b36f48b90930714accf03b411b60d5d948e5/grbl/cpu_map.h#L29 map those pins to Port A, PA2,PA3, PA4 on the Arduino Mega's pins 24,25,26, etc ...

It looks like one could use the V3 shield's double-row of connectors for XSTEP/DIR...ASTEP/DIR with jumpers to the MEGA's extended pins.  grbl doesn't do 4 axes of steppers, so it's these 9 wires up to the CNC Shield V3 for control:

    Mega 13 >>  EN   |  GND
    Mega 24 >> XSTEP | XDIR << Mega 30
    Mega 25 >> YSTEP | YDIR << Mega 31
    Mega 26 >> ZSTEP | ZDIR << Mega 32
               ASTEP | ADIR 
    Mega 5V >>    5V | GND  << Mega GND
     
Then bypass the board and hook your X,Y,Z stops up as 10,11,12; and RESET, HOLD, RESUME, DOOR as A8,A9, A10,A11, and the spindle EN,DIR as 6,7, and the coolant Flood/Mist as 8,9.

The CNC Shield V3 won't work neatly attached to the Arduino Mega.