gnea/grbl-Mega #111: Feature Request : add support for torch up torch down THC and  torch step torch dir inputs

sinfocomp, 2019-09-01:
it would be nice to have two pins for thc control for plasma users, configurable as either torch up torch down inputs, (moving the torch up or down while running program)
or as a torch step dir input (moving sinle step per pulse again while executing a program)
a third desirable function would be sonething like, hold on spindle enable ( so we can use the resume input as an "arc good" input
Thanks

bdurbrow, 2019-09-04:
What are you using as a source of torch height information? Another Arduino... or something else entirely?

:)

sinfocomp, 2019-09-04:
Hello, I'm developing both a voltage divider height control for plasma and also a capacitative one for oxifuel

sinfocomp, 2019-12-15:
These would be the most desirable enhancements to GRBL mega:[http://linuxcnc.org/docs/devel/html/plasma/plasmac-user-guide.html#cha:plasmac-user-guide](url)
thanks