gnea/grbl-Mega #113: Update to gnea/grbl v1.1h (2019-08-25) Release

plainolddave, 2019-11-02:
Hi there - is there a plan to align grbl-Mega with gnea/grbl (as opposed to grbl/grbl) please?

The reason is I just noticed there was a new release of gnea/grbl in August 19, so thought it may be easier to maintain if grbl-Mega was forked from this newer version from the gnea repo - appreciate your thoughts and guidance on whats the way ahead for gnea/grbl-Mega

Thanks

bdurbrow, 2019-11-03:
I'm sure I'll get corrected here if I'm wrong (and @chamnit, please do 😄 ) but I don't think that there is any plan for anything other than severe bug fixes for Grbl-Mega... and I don't think that there are any _severe_ bugs (a few _questionable_ spots, but nothing severe).

The reason for that is most if not all of @chamnit's development effort is going into the ARM version: https://github.com/gnea/grbl/issues/67

However; there are several 3rd party forks of Grbl-Mega that are pushing Grbl as far as we can go on the AVR platform - I've got a version that implements a UI on common 3D printer hardware, @fschill  and @HuubBuis have been working on spindle-sync for lathes (among other things), and @fra589 has a 4/5 axis version; just to name a few.

https://github.com/bdurbrow/grbl-Mega
https://github.com/fschill/grbl-Mega
https://github.com/HuubBuis/grbl-L-Mega
https://github.com/fra589/grbl-Mega-5X

chamnit, 2019-11-03:
For the most part, that's correct. Finding free time has been quite difficult over the last year. I've been re-arranging my priorities to better suit the community in the long run. Having these other forks/ports available help with devoting more time into the difficult stuff.

plainolddave, 2019-11-03:
> For the most part, that's correct. Finding free time has been quite difficult over the last year. I've been re-arranging my priorities to better suit the community in the long run. Having these other forks/ports available help with devoting more time into the difficult stuff.

Completely understand - we don't want to distract you at all!

Nice to keep them in sync though... would you be open to creating a 'maintenance' branch for grbl-Mega - that way we can do the work to align code-bases so won't take your time, and won't effect the main grbl-Mega trunk unless/until its merged sometime.....

Intent from my perspective is just to help out with a little code maintenance if you want - seems cleaner to keep the codebase here as opposed to yet-another-fork. Let me know if that works you (or if not then I understand and close this issue) - thanks again for all your hard work