gnea/grbl-Mega #114: Question Help wanted

sinfocomp, 2019-11-23:
can some one please explain how this works ?

  // Set Timer up to use TIMER4B which is attached to Digital Pin 8 - Ramps 1.4 12v output with heat sink
  #define SPINDLE_PWM_MAX_VALUE     1024.0 // Translates to about 1.9 kHz PWM frequency at 1/8 prescaler
  #ifndef SPINDLE_PWM_MIN_VALUE
  #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
  #endif
  #define SPINDLE_PWM_OFF_VALUE     0
  #define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)

  //Control Digital Pin 6 which is Servo 2 signal pin on Ramps 1.4 board
  #define SPINDLE_TCCRA_REGISTER    TCCR4A
  #define SPINDLE_TCCRB_REGISTER    TCCR4B
  #define SPINDLE_OCR_REGISTER      OCR4C
  #define SPINDLE_COMB_BIT          COM4C1

  // 1/8 Prescaler, 16-bit Fast PWM mode
  #define SPINDLE_TCCRA_INIT_MASK ((1<<WGM40) | (1<<WGM41))
  #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))
  #define SPINDLE_OCRA_REGISTER   OCR4A // 16-bit Fast PWM mode requires top reset value stored here.
  #define SPINDLE_OCRA_TOP_VALUE  0x0400 // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

  // Define spindle output pins.
  #define SPINDLE_PWM_DDR   DDRH
  #define SPINDLE_PWM_PORT  PORTH
  #define SPINDLE_PWM_BIT   5 // MEGA2560 Digital Pin 8

fra589, 2019-11-23:
Perhaps you do a little bit search and you should see other (olders) issues before posting new question...
The answer is often present : https://github.com/gnea/grbl-Mega/issues/55#issuecomment-403261212

sinfocomp, 2019-11-23:
Hello @fra589 , Im not facing that issue, i wish to learn how these settings work, since i need a 1khz and 500hz PWM output to suit the requirements of my laser psu, the way it is now (Im guessing a too high frequency) it does not allow for the plasma discharge to be stable, causing that after 80% pwm the power actually drops instead of going higher, i did not had these issue with GRBL on 328P processor, and for now im setting my laser power manually via pottentiometer, so i was seeking some tuition on how these configuration settings worked so i can learn how to adjust acordingly, Thanks and great Port

fra589, 2019-11-24:
Hello @sinfocomp,  
Oh sorry... Your need was not so explicit for me :smirk:  
Perhaps you will find reply about PWM ferquency managment in Arduino here:  
https://forum.arduino.cc/index.php?topic=72092.msg541587#msg541587
Then, go to the spindle_control.c file, in spindle_init() function to make your adjustments.
@++;  
Gauthier.

sinfocomp, 2019-11-24:
Hello, i have read and re-read many times, tryed to set prescaler to 1/32 but i dont get the results i need, can some one please shine a light on how to setup fot 1khz pwm frequency, thanks

bdurbrow, 2019-11-24:
OK, first, you should download and read the ATMega2560 data sheet if you haven't already. In particular, sections 17 and 18.

> since i need a 1khz and 500hz PWM output

I'm not quite sure what you mean by that... 1Khz frequency, 50% duty cycle? Or you want the settings for both a 1Khz frequency and a 500hz frequency?

The timer is configured with these two lines:

```
#define SPINDLE_TCCRA_INIT_MASK ((1<<WGM40) | (1<<WGM41))
#define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))
```

The WGM bits set up the Waveform Generation Mode, and the 4 in the WGM bits indicates timer 4. The other number indicates which bit of the Waveform Generation Mode the constant is. These bits are described in Table 17-2. 

The CS bits setup the prescaler; these are described in Table 17-6.

To get a 1Khz PWM frequency, these should work:

```
#define SPINDLE_PWM_MAX_VALUE 2000.0

#define SPINDLE_TCCRA_INIT_MASK ((1<<WGM40) | (1<<WGM41))
#define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))

#define SPINDLE_OCRA_TOP_VALUE 2000
```

To get a 500hz frequency, just double the SPINDLE_PWM_MAX_VALUE and SPINDLE_OCRA_TOP_VALUE values:

```
#define SPINDLE_PWM_MAX_VALUE 4000.0
#define SPINDLE_OCRA_TOP_VALUE 4000
```

These values were calculated using this:
https://eleccelerator.com/avr-timer-calculator/

(no affiliation, it's just the first thing that popped up in Google).

Note that I have not tested this, I'm sitting in a restaurant right now and don't have my o-scope or an Arduino with me.

sinfocomp, 2019-11-25:
@bdurbrow , Thanks for the help, im almost there, i changed my config.h as follows
Pin 8 - Ramps 1.4 12v output with heat sink
  #define SPINDLE_PWM_MAX_VALUE     256.0 // Translates to about 1.9 kHz PWM frequency at 1/8 prescaler
  #ifndef SPINDLE_PWM_MIN_VALUE
  #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
  #endif
  #define SPINDLE_PWM_OFF_VALUE     0
  #define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)

  //Control Digital Pin 6 which is Servo 2 signal pin on Ramps 1.4 board
  #define SPINDLE_TCCRA_REGISTER    TCCR4A
  #define SPINDLE_TCCRB_REGISTER    TCCR4B
  #define SPINDLE_OCR_REGISTER      OCR4C
  #define SPINDLE_COMB_BIT          COM4C1

   
  //16-bit Fast PWM mode
  #define SPINDLE_TCCRA_INIT_MASK ((1<<WGM40) | (1<<WGM41))
    // 1/8 Prescaler
  #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))

  #define SPINDLE_OCRA_REGISTER   OCR4A // 16-bit Fast PWM mode requires top reset value stored here.
  #define SPINDLE_OCRA_TOP_VALUE  0x0100 // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

  // Define spindle output pins.
  #define SPINDLE_PWM_DDR   DDRH
  #define SPINDLE_PWM_PORT  PORTH
  #define SPINDLE_PWM_BIT   5 // MEGA2560 Digital Pin 8

ii now have a more stable beam (not flickering any more)
however in my grbl setings
my min pwm is 0 and my max pwm is 1000
this so i can easily change my power settings in a 1% increments (ideally)
however if i go past S800 in a Gcode file, the actual laser power starts droping, i know my laser can still go higher becouse my ma meter is reading roughly 17ma and if go to manual power (via pot) i can get upto 25ma any advice on how to get max power on it ?
thanks a lot
Ed

sinfocomp, 2019-11-25:
BTW is the pwm 5v or 3.3v ?

sinfocomp, 2019-11-25:
so after much reading i came across a solution here
https://github.com/gnea/grbl-Mega/issues/55#issuecomment-403261212
i just changed to a 8bit timer and now getting 0 to 5v pwm control at the frequency i needed
closing

closed 2019-11-25 by sinfocomp