gnea/grbl-Mega #118: Problem with firmware and ramps 1.4

RaffaelB, 2020-03-18:
hello, I have installed various firmware for arduino mega 2650 and ramps 1.4, many of these when I launch UGS, they release the error grbl has not finished loading, only one versions I can load and get communicated, but the engines do not respond, where I am wrong ?, thanks for the help, and sorry for my english.

Nandox7, 2020-03-21:
Having the same problem when tried this on RAMPS 1.4 + MEGA.
Loaded the last version .hex file, after I can connect without any problem.
When I send any movement command I can see the reported position changes but there is no movement done in the stepper motors.

Hardware, drivers an steppers are ok because I tested it in a second controller I have running GRBL with a arduino uno shield.

Nandox7, 2020-03-21:
Aha! found the issue.
While comparing the pin mappin gin cpu_map.h noticed the ifdef statements.

Leaving it here for reference, need to comment & uncoment the ones bellow to use the proper pin mapping for the RAMPS 1.4 inside the config.h file.

// Define CPU pin map and default settings.
// NOTE: OEMs can avoid the need to maintain/update the defaults.h and cpu_map.h files and use only
// one configuration file by placing their specific defaults and pin map at the bottom of this file.
// If doing so, simply comment out these two defines and see instructions below.
//#define DEFAULTS_GENERIC
//#define CPU_MAP_2560_INITIAL

// To use with RAMPS 1.4 Board, comment out the above defines and uncomment the next two defines
 #define DEFAULTS_RAMPS_BOARD
 #define CPU_MAP_2560_RAMPS_BOARD

Due to the way we compile grbl this is prone to happen as none of the config is visible.

RaffaelB, 2020-03-22:
Thanks for the information, I check and try

Shahid-lateef, 2020-07-27:
> Aha! found the issue.
> While comparing the pin mappin gin cpu_map.h noticed the ifdef statements.
>
> Leaving it here for reference, need to comment & uncoment the ones bellow to use the proper pin [...]

But how to change this.