gnea/grbl-Mega #119: A clean option for making spindle PWM work on MEGA2560/RAMPS 1.6

d01010101, 2020-03-27:
I have stumped into the same problem as https://github.com/gnea/grbl-Mega/issues/52 but with generic RAMPS 1.6.

After reading the comments, I would ask if this is the case: grbl-Mega can not control the spindle of a plain CNC without applying a relatively heavy patch under a comment "You should not need to touch these variables", a patch which might become incompatible sooner or later and which already points to wrong lines.

On top of that, D8 puts 12V where you can read in several places "spindle pwn" which in turn can fry one of these boards ready for 5V PWM.

The incompatibility shows even in GRBL control applications like Candle, where the slider for UI control has only two extreme positions possible, which change nothing on D8 by the way.

If the above problems can be solved only with the patch in question, I would suggest, if it is not already done, to provide a mode for the combination MEGA2560/Ramps 1.6/CNC which either works out-of-the-box or requires just a simple #define instead of the patch in question.

I guess that this combination is quite common and so possibly it deserves some clean way of handling it.

d01010101, 2020-03-27:
I am sorry, I confused the other patch, it should be https://github.com/gnea/grbl-Mega/issues/55.

By the way, someone else searching just for PWN https://github.com/gnea/grbl-Mega/issues/103. Possibly grbl-Mega is somewhat printer-centric and CNC owners are on their own, but this little aspect of spindle PWN should be easy to fix, I guess? Some #defines like port, frequency, voltage, used by the code below "should not need to touch". 
Or just one #define CNC which makes a sensible default configuration for CNCs.

fra589, 2020-04-07:
Hi @d01010101,  
You are not alone to ask this, new enhancement added to my todo list: https://github.com/fra589/grbl-Mega-5X/issues/112
@++;  
Gauthier.

fra589, 2022-06-24:
New version of PWM outputs pushed today on grbl-Mega-5X:
https://github.com/fra589/grbl-Mega-5X/issues/112#issuecomment-1165787500