gnea/grbl-Mega #123: Laser connection

winch20010, 2020-04-16:
Hello Gauthier, and thanks for this firmware.
I'm actually trying to moving from repetier to Grbl, and actually, the only issue I have is the laser.

I see in the wiki document different things, and in cpu_map other things...
how can I plug my laser to Grbl ??

in my repetier, it uses to be on Digital 6 in ramps 1.4

in your mega5x pinout diagram, laser is supposed to be on D08 ? (spindle pwn). I don't see any reference to the laser on this diagram

in the cpu_map you shared, I see : 
>>
  #define SPINDLE_ENABLE_BIT      3 // MEGA2560 Digital Pin 6
  ...
  #define SPINDLE_PWM_BIT   4 // MEGA2560 Digital Pin 7
<<

which one is for the laser ?

thanks in advance for your help
Sebastien

winch20010, 2020-04-16:
#define SPINDLE_PWM_BIT 4 // MEGA2560 Digital Pin 7
and what does it means ? Digital pin 7 does not exist in RAMPS right ??

fra589, 2020-04-16:
Hi @winch20010,  
  
You are not on my [grbl_Mega-5X](https://github.com/fra589/grbl-Mega-5X) version project page,  but on the original Mega version...  
  
Warning, you mus use definitions in the `#ifdef CPU_MAP_2560_RAMPS_BOARD` section.  
So, the definition is on the line number 384 of the cpu_map.h file : 
`  #define SPINDLE_PWM_BIT   5 // MEGA2560 Digital Pin 8
`
  
@++;
Gauthier.
  
P.S, close this issue and open another one in https://github.com/fra589/grbl-Mega-5X/issues if you need.

winch20010, 2020-04-18:
ahah better understand now :)  sorry for the confusion !

closed 2020-04-18 by winch20010