gnea/grbl-Mega #126: Serial comunication problem

EragonPower, 2020-06-09:
I've recently encountered a problem regarding the serial comunication between my Arduino mega running grbl-mega and my computer: I've connected my Arduino to an USB port, opened the Arduino IDE serial terminal, and saw nothing, no serial comunication occurred nor TX/RX led blinking. I've tried another computer and another Arduino mega but nothing seems to work. Someone knows what courses this issue?

fra589, 2020-06-09:
Hi @EragonPower,  
  
I recently have a similar issue on Windows 10 with an update who break all USB devices :-(  
cf. https://answers.microsoft.com/en-us/windows/forum/all/windows-10-pro-1909-breaks-all-usb-devices/33caa74b-a0bc-4389-a608-804c2099d170  
Perhaps you can search on [Google](https://www.google.com/search?q=windows+10+update+breaks+usb), it seams that it's not the first time this appear...
  
@++;  
Gauthier.

EragonPower, 2020-06-09:
Is not a windows 10 problem, I've tried on a win 10 and a win 7 machine and nothing worked

EragonPower, 2020-06-09:
The problem is that I've tried on a win 10 and a win 7 machine and nothing seems to work, I've even tried writing some code to test to test the serial read and write capability and everything worked, I think that there are some problems with my Arduino and the way that grbl uses the serial comunication

EragonPower, 2020-06-12:
UPDATE: the RX led blinks when i send commands on the serial but nothing comes back

closed 2020-06-12 by EragonPower