gnea/grbl-Mega #129: Error 22

Shahid-lateef, 2020-07-22:
When I try to move X or Y axis it shows error 22 now what is the solution.

langwadt, 2020-07-22:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-response-messages

Shahid-lateef, 2020-07-22:
> https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-response-messages

How to set feed rate

ChristianBrandtner, 2020-07-22:
Read the manual please.
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

Shahid-lateef, 2020-07-23:
> Read the manual please.
> https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

I read manual but not written this issue now tell me how to fix it please.

langwadt, 2020-07-23:
http://linuxcnc.org/docs/html/gcode/other-code.html#sec:set-feed-rate

Shahid-lateef, 2020-07-23:
> http://linuxcnc.org/docs/html/gcode/other-code.html#sec:set-feed-rate

[image]

See my error

fra589, 2020-07-23:
CNC is not trivial and we can't do this for you:
Read the doc and learn Gcode!!! 
http://linuxcnc.org/docs/html/gcode/g-code.html

Shahid-lateef, 2020-07-24:
> CNC is not trivial and we can't do this for you:
> Read the doc and learn Gcode!!!
> http://linuxcnc.org/docs/html/gcode/g-code.html

No send me some settings for running a CNC and after doing these settings  I will change some value according to my setup.

ChristianBrandtner, 2020-07-24:
> > CNC is not trivial and we can't do this for you:
> > Read the doc and learn Gcode!!!
> > http://linuxcnc.org/docs/html/gcode/g-code.html
> 
> No send me some settings for running a CNC and after doing these settings I will change some value according to my setup.

No,
use the GRBL defaultsettings (use $RST command, for more information see https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands) and then change some values according to your setup.
Learning by doing,  that's what you have do do!

Shahid-lateef, 2020-07-24:
> No,
> use the GRBL defaultsettings (use $RST command, for more information see https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands) and then change some values according to your setup.
> Learning by [...]

What is the full command of feed rate I know G93 and G94 is cmd of feed rate but how to set feed rate.

langwadt, 2020-07-24:
no it is not a command, did you not read the links provided?
G94 alone set the feedrate type to mm(inches) per minute (you don't need it, it is default unless you have changed it)
F500 alone or at the end of line with a move sets the feed rate to 500mm/minute

Shahid-lateef, 2020-07-24:
> no it is not a command, did you not read the links provided?
> G94 alone set the feedrate type to mm(inches) per minute (you don't need it, it is default unless you have changed it)
> F500 alone or at the end of line with a move sets the feed rate to 500mm/minute

Yes I can write these cmd also but error 22 not gone

langwadt, 2020-07-24:
you posted a screen shot, there is no error22 after F260

Shahid-lateef, 2020-07-24:
> you posted a screen shot, there is no error22 after F260

That's only for z axis

langwadt, 2020-07-24:
try again, you only tried x and y before you set the feedrate

Shahid-lateef, 2020-07-24:
No I tried after setting the feedrate

Shahid-lateef, 2020-07-24:
> try again, you only tried x and y before you set the feedrate

No I tried after setting the feedrate also but shows error 22

langwadt, 2020-07-24:
no you didn't because that is not how it works, set the feedrate and it will be the feedrate for everything until you set a new feedrate

Shahid-lateef, 2020-07-25:
> no you didn't because that is not how it works, set the feedrate and it will be the feedrate for everything until you set a new feedrate

I don't know why this happen what do you think what is the problem.