gnea/grbl-Mega #139: Linux sender with more than 3 axis support

Storen3, 2021-01-31:
The fact that Arduino Mega2560 can run up to 6 axes is very exciting. I noticed there are several forks of Arduino Mega for that. However, are there any actively developed G-Code senders for Linux with more than 3 axes?

fra589, 2021-02-01:
Hi @Storen3,  
  
You can try cn5X++ (https://github.com/fra589/cn5X).  
It's the sender I currently develop for my gbbl-Mega-5X version witch is able to drive up to 6 axis.  
cn5X ++ is written in Python under Linux (Debian). It work same under Linux or Windows (perhaps also Mac?). It is in active development and still in beta, but I already use it for real milling.  
  
@++;  
Gauthier.

Storen3, 2021-02-01:
That is great! Thank you very much for working on this.

winder, 2021-02-24:
We're working on ABC support for UGS, it's still in progress but we're getting close:
https://github.com/winder/Universal-G-Code-Sender/pull/1540