gnea/grbl-Mega #142: Is this dead or will there be an update to 1.1h?

d4mer, 2021-04-15:
Hi All, 

Just wondering if this Is dead or will there be an update to 1.1h?

marlboro, 2021-07-06:
I guess from the lack of response this must be dead huh?

fra589, 2021-07-07:
Hi all,  
  
[grbl-Mega-5X](https://github.com/fra589/grbl-Mega-5X),direct fork of grbl-Mega is still active and maintained.  
Last version from may, 10 is 1.1t.  
  
@++;

marlboro, 2021-07-07:
@fra589 Thank you for the heads up! That is exactly the kind of advice I was looking to find.

ccwtruck, 2024-06-14:
> Hi all,
> 
> [grbl-Mega-5X](https://github.com/fra589/grbl-Mega-5X),direct fork of grbl-Mega is still active and maintained. Last version from may, 10 is 1.1t.
> 
> @++;

Still does not support homing on dual-Y (or dual-Y axis at all), Auto-squaring, dual limit switches on each axis -- all features I need.