gnea/grbl-Mega #143: Any sence for ENABLE_SAFETY_DOOR_INPUT_PIN ?

mzeddd, 2021-04-24:
I see the following lines in config.h

```
// This option causes the feed hold input to act as a safety door switch. A safety door, when triggered,
// immediately forces a feed hold and then safely de-energizes the machine. Resuming is blocked until
// the safety door is re-engaged. When it is, Grbl will re-energize the machine and then resume on the
// previous tool path, as if nothing happened.
#define ENABLE_SAFETY_DOOR_INPUT_PIN // Default disabled. Uncomment to enable.
```
But ENABLE_SAFETY_DOOR_INPUT_PIN is not used in other files. It is useless.

At the same time I have this kind of message

> Grbl 1.1r ['$' for help]
> [MSG:Check Door]

It appears every time when I connect MEGA board with RAMPS 1.6 on top of it. This error not happens when I connect only MEGA without RAMPS on it.

What should I do to get it fixed?

When RAMPS on top of MEGA in console it reacts only on "?" command. It does nothing when I enter "$$", "~", "$#" and so on.

RabbitMountain, 2021-08-03:
Have you tried commenting it out if you are not using this feature?

from this #define ENABLE_SAFETY_DOOR_INPUT_PIN // Default disabled. Uncomment to enable.

to this:

// #define ENABLE_SAFETY_DOOR_INPUT_PIN // Default disabled. Uncomment to enable.

mzeddd, 2021-09-07:
If you try o search "ENABLE_SAFETY_DOOR_INPUT_PIN" in all *.h and *.c files you will find it only once in config.h file.

It means that this definition is not used in the code and does not have any sence. It can be commented/uncommented/renamed and nothing happens.