gnea/grbl-Mega #154: Circuits diagram for lcd and Push buttons

MohammadSalehi72, 2022-02-16:
Hi, 
This is an amazing project. I'm trying to implement this project for my proxxon milling machine. Could anyone please share the circuit diagram for "LCD" and "Push buttons" with me? 

Best regards.

fra589, 2022-02-16:
Hi @MohammadSalehi72,  
  
grbl-Mega doesn't support LCD with Push buttons.
For other directly connected to grbl-Mega buttons (other than those with LCD shield), ask Google for "[grbl mega pinout](https://www.google.com/search?client=firefox-b-d&q=grbl+mega+pinout)"

@++;
Gauthier.

MohammadSalehi72, 2022-02-16:
> Hi @MohammadSalehi72,
> 
> grbl-Mega doesn't support LCD with Push buttons. For other directly connected to grbl-Mega buttons (other than those with LCD shield), ask Google for "[grbl mega pinout](https://www.google.com/search?client=firefox-b-d&q=grbl+mega+pinout)"
> 
> @++; Gauthier.

Thanks for your answer.
Actually I'm trying to build a stand alone grbl command sender( without need to connect to pc). I found similar project on YouTube (https://github.com/ludwigpetrosyan/grbl-Mega/tree/edge_lcd). It seems this is used grbl library for Arduino mega. I wanted to know about the pushbutton and lcd connection of this project. Could you please help me?

fra589, 2022-02-16:
Sory I can't... But ask to the https://github.com/ludwigpetrosyan/grbl-Mega/tree/edge_lcd project!