gnea/grbl-Mega #155: Error while uploading through Arduino IDE

neerajbagi, 2022-03-03:
Arduino: 1.8.15 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

grblUpload:27:10: fatal error: grbl.h: No such file or directory

 #include <grbl.h>

          ^~~~~~~~

compilation terminated.

exit status 1

grbl.h: No such file or directory

Invalid library found in C:\Users\niraj\OneDrive\Documents\Arduino\libraries\GRBL: no headers files (.h) found in C:\Users\niraj\OneDrive\Documents\Arduino\libraries\GRBL

fra589, 2022-03-03:
Hi @neerajbagi,  
  
The compiling procedure is normally working everywhere. You probably have missed a detail in the compiling procedure: https://github.com/gnea/grbl/wiki/Compiling-Grbl.  
You should have all the grbl files (*.c and *.h) in the "C:\Users\niraj\OneDrive\Documents\Arduino\libraries\GRBL" directory.  
  
@++;  
Gauthier.

neerajbagi, 2022-03-03:
> Hi @neerajbagi,
>
> The compiling procedure is normally working everywhere. You probably have missed a detail in the compiling procedure: https://github.com/gnea/grbl/wiki/Compiling-Grbl. You should [...]

Hi Gauthier!

Thank you for your reply. 

However, I was following the correct procedure for the installation of the grbl library. However, after several attempts, I have figured out the problem. 

Only the 'grbl' folder within the Zipped source code package needs to be copied to the 'Documents/Arduino/Libraries' folder. If you try to unzip the entire Source code package into the 'Libraries', then Arduino does not recognize the library. 

I think this was pretty fundamental. Since I am a beginner, I am learning it now. 

Thanks for the help.

closed 2022-03-03 by neerajbagi

fra589, 2022-03-03:
This is the one precision detail you missed as it is written in the compilation procedure:  
  
[image]

neerajbagi, 2022-03-03:
Yes. I missed that detail.

Anyway, thank you.